T-Troll/alienfx-tools #364: No CLI's work.

davies-w, 2023-07-16:
**Describe the bug**
I've tried running as administrator, normal etc. Powershell, normal shell, just clicking on CLI.  Nothing works, it just brings up a dialog saying "you want to run this", and then it becomes grey screen and then just exits. Nothing works. I've tried alienfan-cli maxrpm, nothing.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
I'd expect to see some text in my shell window

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
 - Model: [e.g. AlienWare m15R3]
 - Model details: [e.g. per-key rgb keyboard]
 - OS: [e.g. Windows 11 stable]

**Additional context**
Add any other context about the problem here.

T-Troll, 2023-07-17:
It requires Administrator CMD.
If you don't have it will ask for elevation (prompt) then open NEW CMD which will be closed immediately after app finish.

So open Administrator CMD, navigate to the tools folder, run from it.

davies-w, 2023-07-18:
That works!!! Thank you so much!!

closed 2023-07-18 by davies-w