T-Troll/alienfx-tools #370: LOTS of Issues/Questions with Fans on X17 R2

TheSQLGuru, 2023-08-18:
I am now on an X17 R2. Dell FINALLY accepted that my previous X17 R1 was a total lemon and replaced it. I have a 12900HK, 3080ti. 

Note: My BIOS setting for the fan is on whatever is the performance setting, in case that has something to do with functionality. Actually, does it??

1) I click Check Max Boost. Speed is adjusted. When the process is over, the fan remains at the maximum speed indefinitely. I have to take some action such as selecting a different Power Mode to get it to reduce back down. All fans have this behavior.

2) It seems like the way the dots and lines on the chart is (eventually) representative of the fan's final 0-100 "curve"? If so, the picture was taken right as the GPU 4 fan was finishing its run. I would have expected it to clear out those points at upper right where the fan speed went up and down by a hundred several times and just have a mostly-45 deg line from lower left to the upper right corner. 

3) Also, sometimes I don't see that process on the chart. What's the story there?

4) Is there a reason for a HUGE difference in max speeds between fans of the same type? GPU 2 maxed out at 4200. GPU 4 hit 6300 but stabilized at 6200. Actually CPU 1 and GPU 2 are both completely unboosted according to the chart verbiage at the top. :-(

5) What does the Default Boost button do?

6) Sometimes the boost check oscillates around the max speed for an extended time (approx 1.5-2 minutes) before deciding to finalize the speed. Is this expected?

7) I am finding that the speed of the fan response seems quite slow. I can run ThrottleStop's 120M benchmark all the way through, with 2 cores at 100deg C, and the fans just start to come up off of 0 as the run is completing 6.x seconds after starting. It doesn't matter what Power Mode I am on (although I use my custom manual curves pretty much all of the time).

8) Speaking of Power Mode, mine now have names of Level NNN, where NNN is between 160 and 171. I don't notice any difference in fan operations between each of the levels. 

9) I do still rarely get the 5 second "freeze" of my computer when I switch Power Modes. 

10) In Manual Power Mode, it seems that unless I have the temperature sensor checkbox selected where I made my curve, the fan does not boost up to the boost speed. That seems not right.

11) Can I have multiple curves for a given fan on different temperature sensors?

12) Do you know where I can find information regarding the new ESIF sensor 1 .. 8 temperature sensors this X17 R2 has? 2 of them zoom to 100C when I am benchmarking, so I am quite curious about these sensors, and Google doesn't provide any info that I could find.

TheSQLGuru, 2023-08-18:
UPDATE: during all of the above, my fan BIOS setting was on normal, NOT performance

T-Troll, 2023-08-19:
1. Should be reset to normal. I'll check.
2. It's OK.
3. Can you explain?
4. BIOS PWM settings (and, in fact, different fans). All AW's have this.
5. Reset boost to 100 and max. RPM to BIOS-defined value
6. Yes. Fans are imbalance, so it makes sure the result is stable.
7. Yes, fans are inertial. Any fans. I take special measures for 100+ boost (yes, by delay over 100 switch) - otherwise it can take minutes to spin up. BTW, it's also a reason to keep fans running all the time at 5-10% boost - spin up from stop is SLOW.
8. But it is. Also check PLs for CPU and GPU. You can rename it later according to the results.
9. So BIOS hang some operation. Need the BIOS dump to check.
10. If no curves assigned for fan, it stays BIOS-controlled with 0 boost. It's common operation mode.
11. Yes. It's one of the biggest bonus of this app. F.e, i use GPU fan to cool down mine SSD.
12. It should be filled automatically IF you have DPTF drivers (Intel Thermal Framework) installed and files copied (see readme). In case it fails - you can repeat check - stop app, go to registry, change `DPTF` key to 1 and start it again. This will initiate a new check (can take up to 6 seconds!)

TheSQLGuru, 2023-09-17:
Sorry for huge delay. :(
[LOTS of Issues and Questions with Fans on X17 R2 20230818.docx](https://github.com/T-Troll/alienfx-tools/files/12643058/LOTS.of.Issues.and.Questions.with.Fans.on.X17.R2.20230818.docx)

TheSQLGuru, 2023-09-17:
BTW, I wiped my prior installation and installed a fresh 8.4.1 using .msi file. I did Check Max. Boost on CPU 1 fan. It went to 4300, said the check was done, and the fan stayed at 4300 indefinitely (I changed PL to get it to release from max and taper back down to steady-state).

Ahhh! After some permutations, I have isolated the cause of the fan rpm hanging at the top of the rang during check max. It happens when I am on Manual power level. Any other PL, and the fan returns back to baseline.

For #12 (DPTF causing ESIF... names to resolve), I have installed DPTF and Innovation Platform stuff, and I still have ESIF. Perhaps a reboot will fix that??

T-Troll, 2023-09-20:
> and the fan stayed at 4300 indefinitely

Got it, it's a bug (i change fan control under the hood and forget to reset some). I'll fix this.

> For https://github.com/T-Troll/alienfx-tools/issues/12 (DPTF causing ESIF... names to resolve)

Quit app, open regedit, change HKCU/Software/Alienfan/DPTF to 1, start app. Be ready to wait up to 6 seconds for app start. In case you still don't have names - i'll give you the other way to check what is going wrong.

T-Troll, 2023-09-20:
> Sorry for huge delay. :( [LOTS of Issues and Questions with Fans on X17 R2 20230818.docx](https://github.com/T-Troll/alienfx-tools/files/12643058/LOTS.of.Issues.and.Questions.with.Fans.on.X17.R2.20230818.docx)

I will be huge reply, please give me some time.