T-Troll/alienfx-tools #371: Alien Fan Badness

TheSQLGuru, 2023-08-19:
X17 R2 12900HK, 3080, AFX v8.4

Laptop had been running a virus check with Alien Fans set to Manual using my curves, BIOS Fan setting to performance, Power to Bitsum Higest Performance, ThrottleStop max CPU perf settings, using a 230W power supply from an old Alienware Laptop.

Changed power config to battery, TS to ultra power savings. Unplugged laptop and moved it to my office and plugged in the 300W power supply. Walked away. When I returned some time later, Alien Fan Control had somehow dropped to ultra-low fan speeds (something like 600, 400, 1900, 500). Entire laptop was HOT soaked (most temps 75C, and it felt like most of the laptop - especially the metal parts of the base, were immediately painful to the touch), many heat-related alarms/warnings were on in TS. 

I don't know if there is anything still available for diagnosis/troubleshooting. But let me know if I can send anything to assist.

P.S. I hope you have been well in my absence. My health is finally improving enough for me to start being active again. I hope it stays that way!!

T-Troll, 2023-09-20:
Well... I'm badly occupied into my main duty this time, but still continue.
Glad to hear you back!

Looks really strange - i can not block the fans from spinning, so it should run up even for the flat curve at 75C. The only thing i can imagine is overboost, then so high boost prevents fan from running (i meet this for some hardware).

So i need reading... Scale, boost level, etc then it happened.

T-Troll, 2023-09-25:
I have other guy to suffer the same issue, but he have G15.
We examine this, and.... this is BIOS issue - system stop react to boost/temp changes completely!
We are found to override - just run `alienfan-cli unlock` and it will return back to normal.
BTW, can you run `alienfan-cli rpm` and send me you SystemID (it's into brackets at the first line)?

TheSQLGuru, 2023-10-14:
I am sorry for the delay in responding.

Is there any further documentation on what, exactly, `alienfan-cli unlock` does? Is the only time I need to run that when the fans stop responding - or are there other instances where it would be beneficial?

D:\Program Files\Utilities\AlienFXTools>alienfan-cli rpm
AlienFan-CLI v8.5.0
Supported hardware (4800) detected, 4 fans, 2 sensors, 7 power states.
Fan CPU 0 RPM: 2000
Fan GPU 1 RPM: 2000
Fan CPU 2 RPM: 1900
Fan GPU 3 RPM: 1900

T-Troll, 2023-10-14:
Aha, 4800. Seems like the same BIOS module.

`unlock` command select manual power mode. Equal to `setpower=0`.

TheSQLGuru, 2023-10-14:
I am running a quite old BIOS for the X17 R2 because Dell locked up most of the CPU tweaking after the version I have. WTF?? This is a GAMING COMPUTER, from a brand that (used to) PRIDE itself on top-tier performance. Sigh...

T-Troll, 2023-10-14:
Offtopic a bit, i have my own teeths sharped about some things in Dell.
I mean, you and other guys with G15 have THE SAME SYSTEMID! And something broken into this BIOS.