T-Troll/alienfx-tools #373: Libre Hardware Monitor sensors not showing anymore after update

DinnerBug, 2023-08-22:
**Describe the bug**
Launching Libre Hardware Monitor before AlienFX-GUI does not result in the program showing as many sensors as before upgrading to the latest version. I'm particularly interested in the GPU core and the CPU core temperatures. Launching Libre Hardware Monitor causes some sensors to appear but not all and not the ones I need. The issue could have also been caused by the latest Windows update(?)

**Screenshots**
[image]

**System:**
 - Model: Alienware Aurora R8
 - Model details: [e.g. per-key rgb keyboard]
 - OS: Windows 10 Home
 
 Thanks

DinnerBug, 2023-09-08:
after updating to the latest version, the issue persists

T-Troll, 2023-09-08:
Oh, i found the source.
Libre now delivers a lot of sensors, but i only take first 32. Now i increase the limit to 256. ^_^

Check this version - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/12558990/alienfx-gui.zip)

I change the method i count it into next release.

closed 2023-09-08 by DinnerBug