T-Troll/alienfx-tools #375: Alienware/Dell AW3423DW monitor not working with v8.4.1

ahrysicos, 2023-09-19:
The navigation joystick and power buttons on my Alienware/Dell AW3423DW monitor no longer function with v8.4.1 installed. LED lighting was not functioning as intended and the monitor itself also went into standby mode on its own a couple of times. When booting up the PC, no signal is detected until I switch off/on the monitor power at the power outlet. Very strange. No issues at all rolling back to v8.4.0. Running the latest Nvidia drivers and the following:

Edition         Windows 11 Enterprise
Version         22H2
OS build        22621.2283

T-Troll, 2023-09-20:
Ohh... I work actively for 34XXDWF, so i can break something in the meanwhile. Thank you for mention, i'll check and may return with questions.
BTW, check Devices&Grids tab at 8.4.1 - what API version is mentioned for it?

T-Troll, 2023-09-23:
Please check this version - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/12706041/alienfx-gui.zip)
Did it fix you issue?

ahrysicos, 2023-09-24:
Sorry about the delayed response. APIv6 listed for 8.4.0 and 8.4.1. No change with your recent fix unfortunately :( Rolled back to 8.4.0 and everything is back to normal working order.

T-Troll, 2023-09-24:
Ok, thank you. Seems like layout is broken....
What about this version?
[alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/12708635/alienfx-gui.zip)

ahrysicos, 2023-09-24:
Should I just overwrite this GUI exe from my existing 8.4.0 and test, or update to 8.4.1, then overwrite with this GUI exe?

T-Troll, 2023-09-24:
You can just unpack it any location and run - my tools don't have dependence of other stuff.

ahrysicos, 2023-09-25:
No luck unfortunately. With this latest exe, a technical diagnostic menu on the left hand side of the monitor for a few seconds, before the monitor switched off. I had to unplug and plug back in to get it going again.

T-Troll, 2023-09-25:
Yep. Seems like i found the source of the issue (it also affects V9).
Check this - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/12712047/alienfx-gui.zip)

ahrysicos, 2023-09-25:
No luck again unfortunately. All LED lights on the monitor flash repeatedly, then the technical diagnostic menu appears. I did notice when hovering over the AlienFX icon in the system tray, that Effects usually says "Ambient" next to it, based on my preset. However, with these later non working versions, it's listed as "mAhg", whatever that means. I hope that helps.

T-Troll, 2023-09-25:
This means the same - MontoringAmbientHapticsGrid, just shorter.
Aha, so light control back! But it just re-init monitor continuously. Thanks, let me check....

ahrysicos, 2023-09-25:
Ahh, okay. I didn't quite get far enough to determine whether light control was back. It wasn't working while I had it on. Even when I roll back to 8.4.0 I have to go through the process of unplugging the monitor for a little bit, plugging it back in, turning it off and on, etc to get button response and light control back properly.

T-Troll, 2023-09-25:
I found another bug i miss.
Try this version - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/12715458/alienfx-gui.zip)

ahrysicos, 2023-09-26:
Sorry to leave you in suspense for almost a day. You did it! Works flawlessly now with this latest one. Thank you! :)

T-Troll, 2023-09-26:
Great. Thanks for mention it and for help with testing!

closed 2023-09-26 by T-Troll

ahrysicos, 2023-09-26:
Happy to assist!