T-Troll/alienfx-tools #378: Aurora R7

Hardantheweeb, 2023-10-05:
I feel like I've tried everything now but I can't get the FanControl to work at all. Tried both GUI and CLI, with the [8.5.0](https://github.com/T-Troll/alienfx-tools/releases/tag/8.5.0) + SDKv1 and Registry Update. It finds the fans and three sensors, it doesn't seem to save my settings after input through the CLI. 

[image]
[image]

[image]
[image]
[image]

Windows 10 1709.

T-Troll, 2023-10-05:
Fan control only available into Manual mode. Add `unlock` or `setpower=0` before fan operation.

Reshuffling, 2023-10-05:
[image]

T-Troll, 2023-10-05:
No. Into ONE command line - `unlock setfans=100,100,100`
In case getfans still show "-1" after this - this is an issue to examine.

Reshuffling, 2023-10-05:
[image]

T-Troll, 2023-10-05:
Check some services - it can stay even after GUI closed.
According to readings, some interfere and even switch power mode back to auto.

T-Troll, 2023-10-05:
Try to use `alienfan-gui` - it can block this, in case it works - look who switch modes.

Reshuffling, 2023-10-05:
[image]

Fan 3 boost randomly too 400-600 RPM and then just dies to 0 for several seconds. 

Have uninstalled FanControl and will reboot.

Reshuffling, 2023-10-05:
[image]
Nope, nothing seems to happen.

T-Troll, 2023-10-05:
Aha. Control doesn't work.
Let me think a bit and check the code.

You are at v1, right?

Reshuffling, 2023-10-05:
> You are at v1, right?
fan-SDK_V1 yes.

T-Troll, 2023-10-08:
I got the point.
R7 has 2 power modes only, and no manual one.
I use the same control (not SDK!) code for both V1 and V2, so it's always report mode as uncontrolled (that's why you don't have red dot on curve).
I'll fix this into the next version.

Reshuffling, 2023-10-08:
Yeah, setpower=2 seems to work.
[image]

T-Troll, 2023-10-14:
Did you check latest version? I make some changes into v1 SDK for your system.

Reshuffling, 2023-10-21:
I just use the latest version, not using the v1 at all with this command:
alienfan-cli unlock setpower=2 setfans=100,100,100 and it works.

T-Troll, 2023-10-22:
Well....

- V1 needed to control your lights (V2 support it, but doesn't work after all tests).
- Your BIOS did not report current boost (always -1 = error). So current GUI will not work correctly so far.
- You don't need to issue "unlock" into V2 - there is no manual mode (mode 0) at your system. Into V1, i emulate it.

So... In case you only need fan CLI, you can use this. In case you need fan GUI or lights, we need to go a bit deeper and tune it for your system more (v1 or v2).

Reshuffling, 2024-08-24:
I am up to get this solved once for all if you have the time and energy. Let me know what you need.

T-Troll, 2024-08-25:
I can add a patch to fan control GUI (but it can affect other systems), but lights is not solved for v2 (BIOS can not accept data correctly, seems like WMI method have an error in description).

Reshuffling, 2024-08-25:
Well R7 are fun machines, I will just go with the V2 CLI command. Lights are what they are. At least the fans work with the CLI. Marking this as closed.

closed 2024-08-25 by Reshuffling