T-Troll/alienfx-tools #379: M17R5 Keyboard lights doesn't seem to work

gdcdesign, 2023-10-06:
Hello! 
My m17r5 Alienware's laptop keyboard lights doesn't seem to change. The ring, alien head and power button all seem to work fine though. I believe on the guide it said something about per key keyboard needing to do something for it to work?  However I couldn't really understand how to do it (not very good with computers in general). Would you mind providing maybe a step by step process please. I would really appreciate it! Thank you very much

gdcdesign, 2023-10-06:
also, been trying to work on the fans. I'm not sure if its a my laptop issue but when I have the high performance mode on (by pressing f1) it turns on and off repeatedly, along with my caps lock keys? thanks again

gdcdesign, 2023-10-06:
Sorry for writing again but when using it at top performance the fans would sometimes suddenly go quiet and remain at around ~2400? and if I use any other setting its almost always 0?

T-Troll, 2023-10-08:
Well...

1. Looks strange. R5 are supported, and pre-defined settings should work. Can you, please, roll back to 8.4.0 (i have settings into light control after) and check it? In case it will control keyboards, let me know. Hope you are created zone and set into Color tab already.
2. Numlock is just an indicator of Power (Thermal) mode change. It blinks at mode change according to mode number (from 1 to N). 171 (0xAB) is, in fact, "High performance", it's always initiated before switch to G-mode. Let me check, seems like it switch two times in this case.
3. Hmm... It can drop a bit after a minute or two for some systems, but it's not your case. 0 at other modes is OK at low temperature (below 45-50C depends on the system).

JazzyJazzjazzy, 2023-12-20:
> also, been trying to work on the fans. I'm not sure if its a my laptop issue but when I have the high performance mode on (by pressing f1) it turns on and off repeatedly, along with my caps lock keys? Works fine when its the G mode but the light only turns on after putting it on 171 then to g mode, otherwise it shows ~3000ish instead of ~4000 for some reason. I'm so confused right now, hope I'm not troubling you, thanks again

JazzyJazzjazzy, 2023-12-20:
> also, been trying to work on the fans. I'm not sure if its a my laptop issue but when I have the high performance mode on (by pressing f1) it turns on and off repeatedly, along with my caps lock [...]

When i hit F! Key for G-Mode, it flashed continuously when I still had AWCC installed. Once i uninstalled AWCC then the light illuminated steady again.