T-Troll/alienfx-tools #381: AlienFan Control GUI closes in background when closing any game. (M16 R1, WINDOWS 11)

roboboredom, 2023-10-09:
The program I am referring to is "AlienFan Control GUI".

**Describe the bug**
After starting, the program closes silently in the background when closing any game on the computer. It also seems to close randomly after certain amounts of time.

**To Reproduce**
Steps to reproduce the behavior:
1. Install the software using the latest version and the .msi
2. Add firewall and windows defender exceptions for the program
3. Perform all correct configuration steps.
4. Play some steam games and close them.

**Expected behavior**
The program continues to run in the background no matter what games are opened or closed.

**System (please complete the following information):**
 - Model: Alienware m16 r1
 - Model details: cherrymx keyboard
 - OS: Windows 11 stable

T-Troll, 2023-10-14:
It is not closed, in fact, but stop with exception.
Seems like you have some configuration issue.
Can you, please:
1. Open command line (CMD) at application folder
2. Issue the command `alienfx-config backup`
3. It will create 3 or 4 .reg files with application configuration inside. Zip it into archive
4. Attach this archive here for investigation.

roboboredom, 2023-10-14:
Nevermind. I apologize. I reinstalled the latest update, and the issue stopped. It seems something was messed up with the registry

T-Troll, 2023-10-14:
Ah, ok.
Older version HAVE an issue with tooltip - so pointing app tray icon can close it.

T-Troll, 2023-10-14:
I'll close it, reopen in case to meet it again.

closed 2023-10-14 by T-Troll