T-Troll/alienfx-tools #391: Is i9-13900HX Supported Yet?

JazzyJazzjazzy, 2023-11-21:
Hello,

I downloaded the latest version and it shows TWO CPUs in the window and only one curve.

Have you written in support for the i9-13900HX yet? Thank you.

T-Troll, 2023-11-22:
1. Share a screenshot, please. Tool is not depend on CPU type, Alienware platform only.
2. Yes, m18R1 supported. But automatic profile is not into database yes, use manual setup.

JazzyJazzjazzy, 2023-12-18:
Apologies for late reply. I suspect  that I only have two CPUs listed because I did not install your files from the DPTF folder. No problem.

I have Alienware 18 with Cherry Dark Keyboard. I attempted to set up my lights but I am not meeting with success. Kindly have a look at my photos.

Nothing happens on the keyboard lights. Am I missing something? AWCC is completely uninstalled. Thank you.

[image]
[image]
[image]

JazzyJazzjazzy, 2023-12-18:
[image]

JazzyJazzjazzy, 2023-12-18:
Also noted that G-Mode Key lights up when depressed. It has no effect, though. The Fan Curve in ALienFan manages the fans. The boost from the Power Profile changing to TURBO does not occur since AWCC is not installed. Basically, only the KEY illuminates. Thoughts?

https://www.dell.com/support/manuals/en-us/alienware-command-center/awcc_ug_6.x/performance-page-g-key-mode-in-system-and-game-mode-for-dell-g-series-computers?guid=guid-f26f92fa-6f76-4e72-b6c6-ac9055d77a1e&lang=en-us#:~:text=Table%20of%20Contents-,Performance%20page%20G%2Dkey%20mode%20in%20system%20and%20game%20mode,activate%20the%20G%2Dkey%20mode.&text=Once%20the%20G%2D%20key%20is,and%20a%20notification%20is%20displayed.

T-Troll, 2023-12-21:
No, you have 4 fans, not 2! 2 for CPU, 2 for GPU and can control all of them.

About keyboard lights - at some devices (f.e. 410k/510k) one key can have 2 IDs - one for on, one for off (f.e. for CapsLock etc).
So check more IDs than it's off - it can be this.

JazzyJazzjazzy, 2023-12-21:
> No, you have 4 fans, not 2! 2 for CPU, 2 for GPU and can control all of them.
> 

Understood, four fans total.

> About keyboard lights - at some devices (f.e. 410k/510k) one key can have 2 IDs - one for on, one for off (f.e. for CapsLock etc). So check more IDs than it's off - it can be this.
> 

How to check IDs?

> And yes, you don't have DPTF (ESIF) working correctly, seems like your DPTF driver doesn't support it. Or maybe Intel renames the sensor block, and we need to locate the new one.

How to locate the name?

JazzyJazzjazzy, 2023-12-21:
In AlienFan, when I select CPU 1 on the Left side, highlight GPU on the Right side, the Modified Curve is displayed.

BUT, when I select GPU on the Left side and highlight CPU on the Right side, an Unmodified CPU Curve is displayed? 

<img width="468" alt="Screenshot 2023-12-2 065855" src="https://gith
[image]
ub.com/T-Troll/alienfx-tools/assets/40358464/4ae97652-b448-47dc-bc2f-a404e0bad318">

T-Troll, 2023-12-23:
> BUT, when I select GPU on the Left side and highlight CPU on the Right side, an Unmodified CPU Curve is displayed?

Curves assigned per-fan, so yes, you can (and should) have different curves for every fan.

> Is it possible to display the Default Bios Fan Curve?

Yes, but useless. It's zero-zero.

JazzyJazzjazzy, 2023-12-23:
Thank you, will do.

closed 2023-12-23 by JazzyJazzjazzy