T-Troll/alienfx-tools #396: W10 22h2: Appcrash, no dialog at all

kuerbis42, 2023-12-06:
**Describe the bug**
The program crashes immediately, only the traybar-icon is visible, but when hitting with the mouse, it disapers

**To Reproduce**
Steps to reproduce the behavior:
1. start alienfx-gui.exe (as admin or not)

Windowsversion: w10 22h2
Model: Alienware Area 51 R2 (2016)

event viewer report:

```
Protokollname: Application
Quelle:        Windows Error Reporting
Datum:         06.12.2023 12:33:36
Ereignis-ID:   1001
Aufgabenkategorie:Keine
Ebene:         Informationen
Schlüsselwörter:Klassisch
Benutzer:      Nicht zutreffend
Computer:      PCPC
Beschreibung:
Fehlerbucket 2178571875590848185, Typ 4
Ereignisname: APPCRASH
Antwort: Nicht verfügbar
CAB-Datei-ID: 0

Problemsignatur:
P1: alienfx-gui.exe
P2: 8.5.2.1
P3: 6561a3ba
P4: alienfx-gui.exe
P5: 8.5.2.1
P6: 6561a3ba
P7: c0000005
P8: 0000000000030f52
P9: 
P10: 

Angefügte Dateien:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERFBAE.tmp.dmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERFC0C.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERFC3C.tmp.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERFC5E.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERFC8E.tmp.txt

Diese Dateien befinden sich möglicherweise hier:
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_alienfx-gui.exe_69b3d9965867eed752b6caa4311356881640ec37_84def678_25b32280-824a-462e-9213-f16e8cc96e8d

Analysesymbol: 
Es wird erneut nach einer Lösung gesucht: 0
Berichts-ID: 22b3aeab-4457-4fb6-bbd2-621e052940d1
Berichtstatus: 268435456
Bucket mit Hash: 02569757e1c752eade3bd7940a3fbeb9
CAB-Datei-Guid: 0
Ereignis-XML:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Windows Error Reporting" />
    <EventID Qualifiers="0">1001</EventID>
    <Version>0</Version>
    <Level>4</Level>
    <Task>0</Task>
    <Opcode>0</Opcode>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2023-12-06T11:33:36.7864792Z" />
    <EventRecordID>96336</EventRecordID>
    <Correlation />
    <Execution ProcessID="0" ThreadID="0" />
    <Channel>Application</Channel>
    <Computer>PCPC</Computer>
    <Security />
  </System>
  <EventData>
    <Data>2178571875590848185</Data>
    <Data>4</Data>
    <Data>APPCRASH</Data>
    <Data>Nicht verfügbar</Data>
    <Data>0</Data>
    <Data>alienfx-gui.exe</Data>
    <Data>8.5.2.1</Data>
    <Data>6561a3ba</Data>
    <Data>alienfx-gui.exe</Data>
    <Data>8.5.2.1</Data>
    <Data>6561a3ba</Data>
    <Data>c0000005</Data>
    <Data>0000000000030f52</Data>
    <Data>
    </Data>
    <Data>
    </Data>
    <Data>
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERFBAE.tmp.dmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERFC0C.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERFC3C.tmp.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERFC5E.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERFC8E.tmp.txt</Data>
    <Data>\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_alienfx-gui.exe_69b3d9965867eed752b6caa4311356881640ec37_84def678_25b32280-824a-462e-9213-f16e8cc96e8d</Data>
    <Data>
    </Data>
    <Data>0</Data>
    <Data>22b3aeab-4457-4fb6-bbd2-621e052940d1</Data>
    <Data>268435456</Data>
    <Data>02569757e1c752eade3bd7940a3fbeb9</Data>
    <Data>0</Data>
  </EventData>
</Event>

Protokollname: Application
Quelle:        Application Error
Datum:         06.12.2023 12:33:35
Ereignis-ID:   1000
Aufgabenkategorie:(100)
Ebene:         Fehler
Schlüsselwörter:Klassisch
Benutzer:      Nicht zutreffend
Computer:      PCPC
Beschreibung:
Name der fehlerhaften Anwendung: alienfx-gui.exe, Version: 8.5.2.1, Zeitstempel: 0x6561a3ba
Name des fehlerhaften Moduls: alienfx-gui.exe, Version: 8.5.2.1, Zeitstempel: 0x6561a3ba
Ausnahmecode: 0xc0000005
Fehleroffset: 0x0000000000030f52
ID des fehlerhaften Prozesses: 0x518
Startzeit der fehlerhaften Anwendung: 0x01da28380c4d4fa2
Pfad der fehlerhaften Anwendung: C:\Users\pcpc\Desktop\aliengui\alienfx-gui.exe
Pfad des fehlerhaften Moduls: C:\Users\pcpc\Desktop\aliengui\alienfx-gui.exe
Berichtskennung: 22b3aeab-4457-4fb6-bbd2-621e052940d1
Vollständiger Name des fehlerhaften Pakets: 
Anwendungs-ID, die relativ zum fehlerhaften Paket ist: 
Ereignis-XML:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" />
    <EventID Qualifiers="0">1000</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>100</Task>
    <Opcode>0</Opcode>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2023-12-06T11:33:35.3003321Z" />
    <EventRecordID>96335</EventRecordID>
    <Correlation />
    <Execution ProcessID="0" ThreadID="0" />
    <Channel>Application</Channel>
    <Computer>PCPC</Computer>
    <Security />
  </System>
  <EventData>
    <Data>alienfx-gui.exe</Data>
    <Data>8.5.2.1</Data>
    <Data>6561a3ba</Data>
    <Data>alienfx-gui.exe</Data>
    <Data>8.5.2.1</Data>
    <Data>6561a3ba</Data>
    <Data>c0000005</Data>
    <Data>0000000000030f52</Data>
    <Data>518</Data>
    <Data>01da28380c4d4fa2</Data>
    <Data>C:\Users\pcpc\Desktop\aliengui\alienfx-gui.exe</Data>
    <Data>C:\Users\pcpc\Desktop\aliengui\alienfx-gui.exe</Data>
    <Data>22b3aeab-4457-4fb6-bbd2-621e052940d1</Data>
    <Data>
    </Data>
    <Data>
    </Data>
  </EventData>
</Event>
```

you may can help.
Greets, Frank

T-Troll, 2023-12-07:
Looks strange, but you are the second person reporting it.
Ok, let's check some basics.

1. What tray icon looks like (alien eye color)?
2. Can you, please, run it and DENY admin elevation?
3. Did you have some configuration before or it's just a first start?

kuerbis42, 2023-12-07:
Hi, thx for the quick answer.

to 3.: nope, nothing. I tried it first with the old AWCC installed, yesterday I kick all old things off and start your tool only.
to 1.: AlienFX Monitor: runs (green Alien), Alienfan-Gui: no run, Fan-Icon (which diapears), Alienfx-Gui: no run, no icon (today) 
I'm not sure, the the last icon, I tried a lot of things.
to 2.: the is no possibility to run it as "normal" user in properties.., DENY "this program make changes..." doesn't run, too.
[image]

greets, Frank

T-Troll, 2023-12-08:
Ohh... If both fan-gui and fx-gui not run this means some issue in fan control init or processing....
Let me think how to diagnose it.
Meanwhile, try to open `\HKEY_CURRENT_USER\SOFTWARE\Alienfxgui` and create DWORD key named `FanControl` with the value 0, then start fx-gui.

T-Troll, 2023-12-09:
Update:
Seems like i know the source of this issue.
Can you please open Administrator CMD and run this command:
`alienfan-cli dump` then share the result?

kuerbis42, 2023-12-12:
Sorry my late answer. 
alienFAN-gui runs, fx.gui not. I'll try to create the dump today.

T-Troll, 2023-12-18:
So, i have the bad news for us - seems like area51 doesn't have the fan detection, but have AW block into BIOS.
Seems like i have some bug which crash it at detection at this case.
Let me make a test tool with the tracing for you to pinpoint and fix this.

About keyboard - press "Detect devices" - i should have map for it (but not sure).
Use "Device" tab to assign light to keys and the grid, then use "colors" tab to assign it to zones and control it.

T-Troll, 2023-12-23:
Fixed in 8.5.2.2.

closed 2023-12-23 by T-Troll