T-Troll/alienfx-tools #397: Aurora R16 Support?

LeeThompson, 2023-12-09:
Is there anyone working on Aurora R16 support?

I have one if I can provide any information.

T-Troll, 2023-12-10:
I glad to hear we have R16 around!

In general, in case Dell doesn't provide something generally new, it should be supported, but will require manual lights setup.

You can try to open Administrator CMD and run this commands to check:
`alienfx-cli probe` - this reveal detected light devices.
`alienfan-cli dump` - this check for the compatible fan control routines and provide BIOS details.

LeeThompson, 2023-12-10:
On the R16, it the probe only affected the ring light, it never affected the other lights.

AlienFan dump is attached.
[alienfan.txt](https://github.com/T-Troll/alienfx-tools/files/13628669/alienfan.txt)

T-Troll, 2023-12-18:
I need "probe" output to check what kind of device you have.
According to dump, fan control should work correctly, you can check GUI or run something like `alienfan-cli rpm` to check.

uberBuilder24, 2024-06-09:
My Aurora R16 is also not working properly. It doesn't recognize all the lights when using the CLI. I also have an Alienware AW920H gaming headset, would I be able to control both from AlienFX Tools?

T-Troll, 2024-06-19:
CLI does not probe all possible IDs, so i recommend make setup in GUI.
Sorry, headset is not supported yet.

teknowledgist, 2024-10-07:
This is what I get on an R16:

````
Microsoft Windows [Version 10.0.22631.4169]
(c) Microsoft Corporation. All rights reserved.

C:\Windows\System32>cd "\Program Files\AlienFX Tools"

C:\Program Files\AlienFX Tools>alienfx-cli.exe /?
alienfx-cli v9.0.1
Dell API not found, 1 devices found.
Unknown command /?
Done.
C:\Program Files\AlienFX Tools>alienfx-cli.exe probe
alienfx-cli v9.0.1
Dell API not found, 1 devices found.
===== Device VID_187c, PID_0550 =====
+++++ Detected as: Alienware AW-ELC, APIv4 +++++
Do you want to set devices and lights names?n
Done.
C:\Program Files\AlienFX Tools>
````

And this is what I see in the GUI (with the "Detected Devices" sub window:
[image]

I only want to control the lighting of the ring and the "alien head" power button.  Let me know if I can do any additional testing or provide any more info.

inunezalvarez, 2024-11-15:
Any update about the support for Aurora R16?

T-Troll, 2024-11-16:
As you can see, light control supported, but you need manual lights setup - no one share settings with me yet.

teknowledgist, 2024-11-20:
@T-Troll What settings are you looking for?  How can I collect them to share?

T-Troll, 2024-11-21:
Read Beginner's guide about manual lights setup (it's not so hard for desktop).
But be careful with power button - better don't play with it (do not use "Power" setting).

brandonkthomas, 2024-12-12:
I got the ring light to work on my R16.

Click "Load Lights and Grids" in the Devices and Grids tab, then create a new Zone in the Colors tab by selecting all 60 LEDs individually.

[R16.csv](https://github.com/user-attachments/files/18114638/R16.csv)

uberBuilder24, 2024-12-12:
> I got the ring light to work on my R16.
> 
> Click "Load Lights and Grids" in the Devices and Grids tab, then create a new Zone in the Colors tab by selecting all 60 LEDs individually.
> 
> [R16.csv](https://github.com/user-attachments/files/18114638/R16.csv)

Here's the thing though, are the lights in order? Because if the lights aren't in order, effects won't work as intended. Do you need to disable AWCC, and does the power light flash for you too?

brandonkthomas, 2024-12-12:
They’re not in order but they can be rearranged. I couldn’t get the power button to react to anything, no.

T-Troll, 2024-12-19:
Flashing power button usually means it's a hardware (but it's strange for desktop), it has 6 pre-defined modes in this case.
You can experiment with "Power" setting at Devices&Grids for it, but be careful!

teknowledgist, 2025-02-05:
@brandonkthomas Thank you for the information and the `csv`.  With that I am able to adjust the "ring" light via the GUI.  

My ultimate (and *simple*) goal is to be able to set a solid and steady color (no effects!) on all the computers in a lab via a script.  I can't go around to each to run the GUI.  After running the GUI and setting up the zone, `alienfx-cli.exe probe` returns all the lights and a "group" (not "zone"), but I can't figure out how to get `alienfx-cli.exe` to change anything.  Every command I try says it completed and no errors, but nothing actually changes.

Running `alienfx-cli.exe probe` on a machine that has never run the GUI, returns only the device, but no lights or "group" (or "zone").  Is there any way to set up the zone without the GUI (i.e. via script)?  

Any assistance with automatically setting the color of the ring is appreciated.

T-Troll, 2025-02-05:
@teknowledgist `alienfx-cli setone=` command do not use zones, but lightID instead. I hope you configure out why.
You can use it even in case you don't ever run GUI.

teknowledgist, 2025-02-05:
Excellent!  This works great as a startup script:

````
FOR /L %%G IN (0,1,61) DO %~dp0alienfx-cli.exe setone=0,%%G,100,0,100
````

I'd love to set the power button color too, but that'll have to be another day.

Thanks!