T-Troll/alienfx-tools #400: alienware x17 r2 support issue,couldn't set to the G-mode

bellyQQ, 2023-12-14:
I didn't find the "G-mode" option in the power mode，then I tried typing FN+F1,and that didnt work too,and i dont understand the other options except the "manual",please check on the screenshot 
[image]

bellyQQ, 2023-12-14:
**System
 - Model: Alienware X17 R2
 - Model details:I9 12900H RTX3080Ti 32Gmem with cherry rgb keyboard
 - OS: Windows 11 Home

T-Troll, 2023-12-18:
This happened if it doesn't have separate G-mode, but "Performance" mode instead.
It should be one of "Level XXX" modes (which is, in fact, "Thermal modes" from AWCC, select it and check.
NB: You can rename "Level XXX" mode after check to reflect what it do.

WilfulTravesty, 2024-08-19:
My Alienware x17 R2 shows the following in the Power menu: Manual, Level 163, Level 160, Level 161, Level 164, Level 165, Level 171. Both Level 171 and Level 164 seem to set the fan to max speed. What do the other levels mean?

T-Troll, 2024-08-20:
171 (0xAB) is the mode set for G-Mode (among other). The rest is model-depended, it's different between hardware. It's, in general, Quiet, Balanced, Cool, Performance in AWCC.
You can get some hints about each mode using HWINFO with some GPU benchmark - it reveals CPU/GPU power limits.

For example, this is PLs for mine m16R1:
[image]

WilfulTravesty, 2024-10-03:
For reference by others, here's what I measured using HWINFO on my Alienware X17 R2 (2022) running 3DMark's Time Spy Extreme, max frame rate off. CPU is an Intel Core i9-12900HK, GPU is an NVIDIA GeForce RTYX 3080 Ti, memory is DDR5-4800.

mode 165 - TSP:201W  CPU Pkg:45W IA Cores:36W  PL1:195/75W  | GPU 125W | Little or no fan speedup (No Fans)
mode 163 - TSP:205W  CPU Pkg:43W IA Cores:35W  PL1:195/75W  | GPU 125W | Mild Fan Speedup (Quiet)
mode 160 - TSP:208W  CPU Pkg:43W IA Cores:35W  PL1:195/165W | GPU 145W | Mild Fan Speedup (Cool)
mode 161 - TSP:210W  CPU Pkg:44W IA Cores:36W  PL1:195/165W | GPU 166W | Fan speeds up with load (Balanced)
mode 164 - TSP:245W  CPU Pkg:51W IA Cores:43W  PL1:195/165W | GPU 164W | Constant high fan speed (Max Fans)
mode 171 - TSP:245W  CPU Pkg:49W IA Cores:43W  PL1:195/165W | GPU 165W | Constant high fan speed (G-Mode)

T-Troll, 2024-10-04:
Nice job!
Guys, can you provide me 2 things:
1. `alienfx-cli dump` output (run it from Administrator CMD!)?
2. BIOS dump (use rweverything to make it)?

I'll check what's wrong with G-mode at your system.

T-Troll, 2024-10-04:
Oh... Sorry... `alienfan-cli dump`

WilfulTravesty, 2024-10-04:
Well, shucks. I tried a few other programs in case you had specified the wrong one, but I must not have tried that program. Below is the output. 

Unfortunately, I've been unable to run rweverything; I tried both the portable and installed versions, and both get stopped by Windows Compatibility Assistant with a dialog that says RwDrv.sys cannot load on this device and was blocked, but turning off Memory Integrity and rebooting didn't make any difference.

C:\Windows\System32>"c:\Program Files\AlienFX Tools\alienfan-cli.exe" dump
AlienFan-CLI v9.0.1
Supported hardware (4800) detected, 4 fans, 12 sensors, 7 power states, XMP.
Names:
[WMI, dynamic: ToInstance, provider("WmiProv"), Locale("MS\\0x409"), Description("WMI Function"), guid("{A70591CE-A997-11DA-B012-B622A1EF5492}")]
class AWCCWmiMethodFunction
{
        [key, read] string InstanceName;
        [read] boolean Active;
        [WmiMethodId(13), Implemented, read, write, Description("Return Overclocking Report.")] void Return_OverclockingReport([out] uint32 argr);
        [WmiMethodId(14), Implemented, read, write, Description("Set OCUIBIOS Control.")] void Set_OCUIBIOSControl([in] uint32 arg2, [out] uint32 argr);
        [WmiMethodId(15), Implemented, read, write, Description("Clear OC FailSafe Flag.")] void Clear_OCFailSafeFlag([out] uint32 argr);
        [WmiMethodId(19), Implemented, read, write, Description("Get Fan Sensors.")] void GetFanSensors([in] uint32 arg2, [out] uint32 argr);
        [WmiMethodId(20), Implemented, read, write, Description("Thermal Information.")] void Thermal_Information([in] uint32 arg2, [out] uint32 argr);
        [WmiMethodId(21), Implemented, read, write, Description("Thermal Control.")] void Thermal_Control([in] uint32 arg2, [out] uint32 argr);
        [WmiMethodId(23), Implemented, read, write, Description("MemoryOCControl.")] void MemoryOCControl([in] uint32 arg2, [out] uint32 argr);
        [WmiMethodId(26), Implemented, read, write, Description("System Information.")] void SystemInformation([in] uint32 arg2, [out] uint32 argr);
        [WmiMethodId(28), Implemented, read, write, Description("Power Information.")] void PowerInformation([in] uint32 arg2, [out] uint32 argr);
        [WmiMethodId(32), Implemented, read, write, Description("FW Update GPIO toggle.")] void FWUpdateGPIOtoggle([in] uint32 arg2, [out] uint32 argr);
        [WmiMethodId(33), Implemented, read, write, Description("Read Total of GPIOs.")] void ReadTotalofGPIOs([out] uint32 argr);
        [WmiMethodId(34), Implemented, read, write, Description("Read GPIO pin Status.")] void ReadGPIOpPinStatus([in] uint32 arg2, [out] uint32 argr);
        [WmiMethodId(35), Implemented, read, write, Description("Read Chassis Color.")] void ReadChassisColor([out] uint32 argr);
        [WmiMethodId(36), Implemented, read, write, Description("Read Platform Properties.")] void ReadPlatformProperties([out] uint32 argr);
        [WmiMethodId(128), Implemented, read, write, Description("Caldera SW installation.")] void CalderaSWInstallation([out] uint32 argr);
        [WmiMethodId(129), Implemented, read, write, Description("Caldera SW is released.")] void CalderaSWReleased([out] uint32 argr);
        [WmiMethodId(130), Implemented, read, write, Description("Caldera Connection Status.")] void CalderaConnectionStatus([in] uint32 arg2, [out] uint32 argr);
        [WmiMethodId(131), Implemented, read, write, Description("Surprise Unplugged Flag Status.")] void SurpriseUnpluggedFlagStatus([out] uint32 argr);
        [WmiMethodId(132), Implemented, read, write, Description("Clear Surprise Unplugged Flag.")] void ClearSurpriseUnpluggedFlag([out] uint32 argr);
        [WmiMethodId(133), Implemented, read, write, Description("Cancel Undock Request.")] void CancelUndockRequest([out] uint32 argr);
        [WmiMethodId(135), Implemented, read, write, Description("Devices in Caldera.")] void DevicesInCaldera([in] uint32 arg2, [out] uint32 argr);
        [WmiMethodId(136), Implemented, read, write, Description("Notify BIOS for SW ready to disconnect Caldera.")] void NotifyBIOSForSWReadyToDisconnectCaldera([out] uint32 argr);
        [WmiMethodId(160), Implemented, read, write, Description("Tobii SW installation.")] void TobiiSWinstallation([out] uint32 argr);
        [WmiMethodId(161), Implemented, read, write, Description("Tobii SW Released.")] void TobiiSWReleased([out] uint32 argr);
        [WmiMethodId(162), Implemented, read, write, Description("Tobii Camera Power Reset.")] void TobiiCameraPowerReset([out] uint32 argr);
        [WmiMethodId(163), Implemented, read, write, Description("Tobii Camera Power On.")] void TobiiCameraPowerOn([out] uint32 argr);
        [WmiMethodId(164), Implemented, read, write, Description("Tobii Camera Power Off.")] void TobiiCameraPowerOff([out] uint32 argr);
};

T-Troll, 2024-10-05:
Yep. You really don't have G-mode support in BIOS, but Performance power mode instead.
Looks strange, but fact.

So it's not my tools bug, just your system design.

closed 2024-10-05 by T-Troll

Pitagoras17, 2024-10-09:
I wanted to give an update on my system (x15 R2 
, i9, 3080 Ti, Latest everything). With the latest AWCC 6.4 update, "Performance Mode" is back. I am able to use AlienFX, AWCC, or the Fn + F1 hotkey to toggle on G-Mode (Performance Mode/Level 171).

Pitagoras17, 2024-10-09:
I wanted to give an update on my system (x15 R2, i9, 3080 Ti, latest everything). With the latest AWCC 6.4 update, "Performance Mode" is back. I am able to use AlienFX, AWCC, or the Fn + F1 hotkey to toggle on G-Mode (Performance Mode/Level 171).

T-Troll, 2024-10-10:
JFYI: Fn+F1 is a software function, you need AWCC or AlienFX to support it (as well as Fn+Alienhead).
But AlienFX map it to "real" g-mode if supported. But i have a shortcut to switch any power mode - CTRL+ALT+0...9.