T-Troll/alienfx-tools #403: 2 Questions

FLASH8ACK, 2023-12-20:
1. I want static green lights ON for my chassies and powerbutton on my Alienware Aurora R8, can this program do this for me?
2. How can i easily make this happen?

Sorry for being such a noob but i have tried and dont understand how the program works.

T-Troll, 2023-12-21:
Uff...
First, let's check what kind of the device you have - run `alienfx-cli probe` from cmd and share output. In the worst case (like for R7) you will need to use R1 version of afx.

It's not so hard, but i don't have ID mappings for your system, so you need to define it manually, as described [here](https://github.com/T-Troll/alienfx-tools/wiki/How-to-start-%28Beginner%27s-guide%29), then you can set it up to any desired color/effect.

FLASH8ACK, 2023-12-21:
C:\Program\alienfx tools>alienfx-cli probe
alienfx-cli v8.5.0
Dell API ready, 3 devices found.
===== Device VID_0461, PID_4ec0 =====
+++++ Detected as: DELL Alienware 610M, APIv7 +++++
===== Device VID_0461, PID_4e9d =====
+++++ Detected as: DELL Alienware 610M, APIv7 +++++
===== Device VID_187c, PID_0550 =====
+++++ Detected as: Alienware AW-ELC, APIv4 +++++

T-Troll, 2023-12-21:
Oh, nice. Seems like i have a small bug with the mouse - it detected 2 times, but it will not prevent you to control it's colors as well.

Fortunately, your device is a common 0x550 chip, so you don't need to do something special - you need just find what light ID responsible for which light control - the process described into the beginner's guide mentioned above.

One thing i need to mention is about power button. At notebooks, power button have special hardware control, but it can be common light into your system (desktop doesn't need to reflect battery power activated ^_^). So then you detect power button ID - check it behavior. In case it just light on - do not set it as a "power button", just name and use. In case it will blink rapidly instead of steady light - it is hardware power button and you need to set checkbox "power" for it.

Good luck!

FLASH8ACK, 2023-12-21:
Couldnt i just use the CLI and "setall=255,0,255"

T-Troll, 2023-12-21:
No. For now, my tools have no idea what "all" lights mean. But you can use `alienfx-cli probe` (answer "y" for question) to set it up into CLI. Read cli usage for more information about this command.

FLASH8ACK, 2023-12-21:
Thanks, I will!

FLASH8ACK, 2023-12-21:
Nah i will prolly throw in the towel and call it a night. I thought i could try to understand it but as i am an mediocre PC user i dont understand it at all. Maybe if the GUI was easier..