T-Troll/alienfx-tools #404: Power Limit Strangeness - Alienware Fan Control

isaacsummers, 2023-12-20:
**Description**
I'm having a strange issue where gaming on High Performance Mode with Alienware Fan Control has some strange power limits.
Both my PL1 and PL2 Static are set to 200W and I have HWINFO64 to see the Dynamic power limits.

When on power mode Level 161 (seems to be my high performance), I notice that PL1 starts at 130W but after 5 or so minutes it drops down to 30W which causes a CPU bottleneck for the GPU. No issues with the Balanced Mode Level 160 as the power limit settles around 55W. The fans definitely are running faster in Level 161.

I would have thought that High Performance Mode would have a higher CPU PL1 so that it can meet higher demanding tasks especially with the fans spun up a bit higher.

**To Reproduce**
Steps to reproduce the behavior:
1. Set Power Mode in Alienware Fan Control to Level 161
2. Stress with GPU usage / game
3. Watch PL1 (Dynamic) drop to 30W over time

**Expected behavior**
PL1 to be higher than where Level 160 settles (55W)

**System (please complete the following information):**
 - Model: Alienware M16R1 - i9 13900hx | RTX 4090
 - Model details: per-key RGB
 - OS: Windows 11 stable

**Additional context**
I'm not running throttlestop or xtu and have uninstalled AWCC. I am running a -125 undervolt on p-cores and -100 cache + e-cores

JazzyJazzjazzy, 2023-12-20:
Unrelated question. May I ask what software you used to UNDERVOLT? I have the same CPU and mine appears to be Locked. Thank you

isaacsummers, 2023-12-21:
> Unrelated question. May I ask what software you used to UNDERVOLT? I have the same CPU and mine appears to be Locked. Thank you

I used https://github.com/DavidS95/Smokeless_UMAF to apply the undervolt at the BIOS level because I need to keep intel virtualization on (allows wsl2). If you don't you can disable core isolation and virtualization and you should be able to undervolt.

JazzyJazzjazzy, 2023-12-21:
I tried SMOKELESS_UMAF and I saw the Option to Disable Undervolt. I selected it and restarted but I still could not enable an Undervolt using Throttlestop.

I have Alienware M18 with BIOS 1.13.0.

T-Troll, 2023-12-21:
@JazzyJazzjazzy It's well-know issue - Dell block undervolt for unknown reason. Roll back BIOS.

@isaacsummers Hmm... Seems like it happened then Turbo timer expired (usually in 90 seconds). This means you PL settings from ThrottleStop not applied correctly. But 30W looks so low for me.
Also, you can rename power mode into app to better understand what is it about. JFYI - digits (like 161) is a mode code into BIOS (0xa1 in your case). Performance mode usually 0xab (171), do you have it into the list?

JazzyJazzjazzy, 2023-12-21:
> When on power mode Level 161 (seems to be my high performance), ...> 

When we change the POWER MODE, should the Power Mode in Windows POWER OPTIONS change to reflect current mode? For example, I am currently in BALANCED mode in Windows and MANUAL MODE in AlienFan. Now, I start changing from MANUAL Mode to the other available Options. Should the Power Mode in windows change respectively?

isaacsummers, 2023-12-22:
> @JazzyJazzjazzy It's well-know issue - Dell block undervolt for unknown reason. Roll back BIOS.
> 
> 
> 
> @isaacsummers Hmm... Seems like it happened then Turbo timer expired (usually in 90 seconds). This means you PL settings from ThrottleStop not applied correctly. But 30W looks so low for me.
> 
> Also, you can rename power mode into app to better understand what is it about. JFYI - digits (like 161) is a mode code into BIOS (0xa1 in your case). Performance mode usually 0xab (171), do you have it into the list?
> 
> PS: I don't mess with PLs any way, i just switch BIOS modes.

Thanks for the response. It's actually when I don't have throttle stop open. If I do it keeps the PL static to what is set (with sync mmio). However if i don't open TS I think it's the Intel dynamic tuning platform that changes the power levels which is strange. Just wasn't expecting such a low PL for being in supposed high performance mode. 

I have renamed each level but haven't checked after the fact going into the bios to see the name. With what I've observed in fan speeds and power level I have

160-balanced
161-high performance
162-quiet
163-cool
164-full speed
165-g mode

Could it be a dell bios issue. I'm currently on their latest.

isaacsummers, 2023-12-22:
> I tried SMOKELESS_UMAF and I saw the Option to Disable Undervolt. I selected it and restarted but I still could not enable an Undervolt using Throttlestop.
> 
> 
> 
> I have Alienware M18 with BIOS 1.13.0.

It will only work if you disable virtualization and core isolation if you go that route. What I've done instead is enable overclocking in dell bios. Restart and boot into smokeless, then change the voltage offsets in there. These apply at boot and allow any range between what is set and up in TS or XTU. 
Note: **make sure you select the offset indicator/prefix to negative**

JazzyJazzjazzy, 2023-12-22:
Aha! Disabled Virtualization & Core Isolation as you suggested and it Worked! Thank you!

JazzyJazzjazzy, 2023-12-22:
160-balanced
161-high performance
162-quiet
163-cool
164-full speed
165-g mode

How can I correlate these values to MY power modes? Is there a CMD that I can use? Thank you.

JazzyJazzjazzy, 2023-12-22:
https://www.minitool.com/news/modern-standby.html

I found the above referencing S0 Power State. S0 removes the other Power Modes; ie. HIGH PERFORMANCE, etc form the Control Panel Power Options. Only BALANCED MODE is available. It appears that we no longer have the other forms of S1, S2, S3. As a result, I will not see Power Mode changes in Windows when I select Mode 161. There is no other Power Plan visible in windows now.

Microsoft Windows [Version 10.0.22631.2861]
(c) Microsoft Corporation. All rights reserved.

C:\Windows\System32>powercfg /a
The following sleep states are available on this system:
    Standby (S0 Low Power Idle) Network Connected

The following sleep states are not available on this system:
    Standby (S1)
        The system firmware does not support this standby state.
        This standby state is disabled when S0 low power idle is supported.

    Standby (S2)
        The system firmware does not support this standby state.
        This standby state is disabled when S0 low power idle is supported.

    Standby (S3)
        The system firmware does not support this standby state.
        This standby state is disabled when S0 low power idle is supported.

    Hibernate
        Hibernation has not been enabled.

    Hybrid Sleep
        Standby (S3) is not available.
        Hibernation is not available.
        The hypervisor does not support this standby state.

    Fast Startup
        Hibernation is not available.

C:\Windows\System32>

T-Troll, 2023-12-23:
First, power modes into app HAVE NO CORRELATION with the Windows Power Plan. It's just proprietary BIOS control settings provided by Dell.

The only exception is CPU MODES - which are, in fact, hidden settings into Power Plan.

Windows have a bud which can delete other power modes. There are a lot of tools around to return it back.

JazzyJazzjazzy, 2023-12-23:
I thought the other Windows Power Plans needed the S1, S2, S3 sleep.statea.available? 

Okay, understand I can recover the other Modes. Thank you.

JazzyJazzjazzy, 2023-12-24:
You can not recover the Power Modes in Windows 11 with a 13th Gen CPU. The required Sleep States are not available nor supported by the CPU. Registry hacks do not work. S0 is the only mode available.

Standby (S3)
    The system firmware does not support this standby state.
    This standby state is disabled when S0 low power idle is supported.

This comment from my powercfg /a command says S3 is not supported. Apparently after 2020, a lot of laptops took away this option. My M18 does not support S1, 2, or 3 in Windows or the BIOS.

isaacsummers, 2023-12-29:
> First, power modes into app HAVE NO CORRELATION with the Windows Power Plan. It's just proprietary BIOS control settings provided by Dell.
> 
> 
> 
> The only exception is CPU MODES - which are, in fact, hidden settings into Power Plan.
> 
> 
> 
> Windows have a bug which can delete other power modes. There are a lot of tools around to return it back.

Does changing the power level in AFX change the CPU modes?

dc-IV-AW, 2024-01-04:
> It will only work if you disable virtualization and core isolation if you go that route. What I've done instead is enable overclocking in dell bios. Restart and boot into smokeless, then change the [...]

Do you know how to deal with "mv Boost" in SmokelessUMAF? Using "mv Boost" usually allows a TS CPU P Cache UV that causes lock ups to function without issues.

isaacsummers, 2024-01-19:
So I've looked into the original issue with the power limits. Still not sure why it throttles to 30W in 161 mode but I believe the issue originates from Intel Dynamic Tuning.

closed 2024-01-19 by isaacsummers