T-Troll/alienfx-tools #405: Macro Keys

mscreations, 2023-12-23:
**Is your feature request related to a problem? Please describe.**
Not really.

**Describe the solution you'd like**
AWCC has control over the 4 macro buttons on my m17 R4. I can't stand AWCC for fan control because it will let the CPU/GPU grenade themselves without kicking the fan speed up unless I manually tell it to go to full speed. (All other modes the fan stay quiet and I end up having massive throttling issues). I've found that this project works great for controlling the fans as I expect them to run so that fixes that issue. I would like to continue using the 4 macro buttons on my keyboard, but I don't know of any other projects that can detect when they are being pressed. Is this something the SDK covers or am I out of luck here?

**System (please complete the following information):**
 - Model: Alienware m17 R4
 - Model details: Per-Key RGB
 - OS: Windows 11

T-Troll, 2023-12-24:
I have macro keys on my keyboard too, but can't RE interface for programming (it's hardware, not just keypress detection).
So the answer is no, it's not supported to date.
But you can use AWCC for it and AFX for fan control - they can co-exist (just turn on "Disable AWCC" option into settings - after this AWCC control will be disabled at AFX start and enabled back at quit).

mscreations, 2023-12-24:
This looks like it's going to require something to monitor the USB device directly which I'm researching. I have found the following information though. I'm posting here in case someone smarter than me knows what to do with the information:

The device appears to be VID: 187c PID:0550 Alienware LED controller Endpoint 1

The keys produce a URB_INTERRUPT in packet that has the pertinent data. I've identified that it generates codes 4C-4F for the 4 keys. On key down (Macro A in this case), the packet HID Data is `02 4C 00` and on key up it is `02 00 00`. It also doesn't appear to matter if any of the Alienware services are running for this data to generate, so I think if I can figure out a way to monitor for these codes, I could do something with it. Still looking to see what needs to be done there.

Either way, I think this would be out of scope for this project so I'll close this issue but thought having this information documented would somewhere would be good.

closed 2023-12-24 by mscreations

mscreations, 2024-01-04:
Revisiting this, I have created a project to satisfy  the macro key requirement. [Alien Macros](https://github.com/mscreations/Alien-Macros). It is *VERY* basic at this point and I've only tested it on my laptop, an Alienware m17 R4. It is mostly proof of concept. If anyone wants to give it a try I'd appreciate input on it.