T-Troll/alienfx-tools #406: AlienFX hangs with Alienware Aurora R12

philkernick, 2024-01-06:
Running AlienFX hangs when selecting the Devices and Grids panel.

Running alienfx-cli.exe probe gives:
alienfx-cli v8.5.0
Dell API not found, 2 devices found.
===== Device VID_0424, PID_274c =====
+++++ Detected as: Microchip Tech Hub Controller, APIv6 +++++
===== Device VID_187c, PID_0550 =====
+++++ Detected as: Alienware AW-ELC, APIv4 +++++
Do you want to set devices and lights names?yes
Found 2 device(s)
Probing device VID_0424, PID_274c, current name Display, #10060, New name (ENTER to skip):
New name Display, #10060
Testing light #0, New name (ENTER to skip):

and at this point the command also hangs.  Ctrl-C to get back to CMD prompt.

**System (please complete the following information):**
 - Model: Alienware Aurora R12 desktop
 - Model details: desktop PC
 - OS: Windows 11

T-Troll, 2024-01-06:
It's a well-known Alienware monitor firmware bug.
Power off your monitor (better remove power cord), then connect it back - it should work correclty after this.

philkernick, 2024-01-06:
I don't have an Alienware monitor.  I have 3 Dell monitors, an S3220DGF and 2x UP2516D, and none of them are RGB capable.

Powercycling all the monitors has fixed the hanging.

However, it now has the Front Ring and Alienhead the wrong way around in the UI.

T-Troll, 2024-01-07:
Aha. Seems like misdetection, they use same chipset like AW one, but RGB blocked/not used.
I even can configure out why it hangs in this situation...
Ok... As a quick fix - remove USB cable(s) from monitor. I'll think about how to handle it correctly into the next version.

T-Troll, 2024-02-24:
Check [8.6.0](https://github.com/T-Troll/alienfx-tools/releases/tag/8.6.0)

philkernick, 2024-02-24:
That works.  No hanging, and it doesn't detect the monitor.

Also the 8.6.0 MSI is flagged as a virus by Microsoft Defender.

T-Troll, 2024-02-24:
> That works. No hanging, and it doesn't detect the monitor.

Great.

> Also the 8.6.0 MSI is flagged as a virus by Microsoft Defender.

Yes, Defender don't like it, and MS don't want to resolve the situation. I even have an explanation into project readme and wiki.
But this software is 100% open source (no binary stubs), so you can download source code, check, and build yourself in case in doubt.