T-Troll/alienfx-tools #407: Config file for X15 r1 please

JeizasK, 2024-01-22:
Any can share config for alienware x15 r1. I am noob and I dont know how to use the app 
[image]
[image]

Justsomebody1234, 2024-02-05:
how did you get X15 r2 to show up ?

T-Troll, 2024-02-19:
Someone share me the configuration.
No one with x15r1 do it yet.... Maybe you?

Justsomebody1234, 2024-02-19:
i'd love to but i have an X15 r2

jcellswo, 2024-07-19:
I'm also looking for a x15r1 config file. I tried to follow the guide for manually setting up lihts and zones, but with no luck. Seems to light up stuff at random...

T-Troll, 2024-08-02:
Light setting is to DEFINE the lights, not CONTROL them.

Unfortunately, there is no way to configure out what lights your gear have, so you can use your own eyes to define it. THEN you can control defined lights.