T-Troll/alienfx-tools #408: Dell G15 5530 AlienFanControl Disable AWCC function not functioning?

Olympus-E, 2024-01-25:
**Describe the bug**
When enabling the "Disable AWCC" Option, AWCC is still running.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
AWCC Should quit automatically

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
 - Model: Dell G15 5530
 - Model details: [e.g. per-key rgb keyboard]
 - OS: Windows 11 Home 22H2 22621.3007

**Additional context**
Add any other context about the problem here.

T-Troll, 2024-02-24:
It not block AWCC UI from running, only stop AWCCService which change colors/fans.

I have 2 suggestions:
- AWCCService not stop (please check into Services then GUI running)
- AWCC 6 has different service name.

Can you check both cases?

Olympus-E, 2024-02-24:
> It not block AWCC UI from running, only stop AWCCService which change colors/fans.
> 
> I have 2 suggestions:
> 
> * AWCCService not stop (please check into Services then GUI running)
> * AWCC 6 has different service name.
> 
> Can you check both cases?

I have AWCC 6.1.11.0

After "Disable AWCC" option is checked:

- AWCC Service is still running (in my case it's Dell Tech Hub service and possibly Fusion Service)

- AWCC processes are still running:
1. AWCC.SCSubAgent.exe
2. AWCC.UCSubAgent.exe
3. AWGameLibrary.SCSubAgent.exe
4. AWGameLibrary.UCSubAgent.exe
5. AWPerformance.SCSubAget.exe
6. AWPerformance.UCSubagent.exe

T-Troll, 2024-02-24:
> But if I disable the Dell Tech Hub service AND restart the computer, the above 6 processes will no longer be present in the task manager after the reboot.

Ok. What happened RIGHT AFTER you stop it (use Manage or Services tab for it)? Also, can you share this service profile screenshot?

Fusion is about power plans control, out of our interest.

Olympus-E, 2024-02-24:
> Ok. What happened RIGHT AFTER you stop it (use Manage or Services tab for it)? Also, can you share this service profile screenshot?
>
> Fusion is about power plans control, out of our interest.

If I stop the service without restarting my computer, nothing will change. All the processes (including above 6) are still running until a restart is performed. (Haven't tried Logging out & back in)

Service Profile: 
[image]

T-Troll, 2024-02-24:
Fortunately, i can read Chinese (but only can speak Cantonese).
I need to see last tab on top (Dependencies).

Olympus-E, 2024-02-24:
> Fortunately, i can read Chinese (but only can speak Cantonese). I need to see last tab on top (Dependencies).

It shows "No Depenencies" :(

[image]

T-Troll, 2024-02-24:
Thanks. I'm going to think about this....