T-Troll/alienfx-tools #411: A Bug about lighting

TZSLLL, 2024-02-08:
Now I can't customize the color of my keyboard light, it seems to be locked.
Since I can't set the lights on my keyboard, I can't reproduce it either, sorry
I'd like to get my keyboard back to normal
This video should explain my problem
My computer is G15 5515
Processor AMD Ryzen 7 5800H with Radeon Graphics 3.20 GHz
System Type 64-bit operating system based on x64 processor
Version Windows 10 Professional
Version number 22H2
Internal OS version 19045.3930

Simply put, after another unexpected stutter, it turned into what it looked like in the video, and I wasn't sure what I'd done in the stutter.
The button in the Setting-Dimming settingsil-Dim lights to adjust the brightness of this area.
When I caused this Bug, I used a Chinese version, the screenshot is attached, but I can't adjust the lighting effect in this area even with the latest version.
I tried to clear the alienfxfan, alienfx_sdk, and alienfxgui folders in the hkey_current_user\software path of the registry, but it didn't work.
Restarting the computer didn't work either.
Should I reset the Bios?
This is my first time using GitHub, I hope you can forgive my mistakes when using this software and Github website.
This article was written in Chinese and translated into English usi

https://github.com/T-Troll/alienfx-tools/assets/159496894/6247c33a-dc58-46d4-b44c-9bf3f6bd5201

ng machine translation.
Below is the Chinese original, you can also use a different machine to translate it to get a better translation.

现在我不能自定义我键盘灯光的颜色，它好象是被锁定了。
由于我的不能设置我的键盘的灯光，我也不能复现它，抱歉
我希望能让我的键盘恢复正常
这段视频应该可以说明我的问题
我的电脑是G15 5515
处理器	AMD Ryzen 7 5800H with Radeon Graphics            3.20 GHz
系统类型	64 位操作系统, 基于 x64 的处理器
版本	Windows 10 专业版
版本号	22H2
操作系统内部版本	19045.3930

简单地说，再一次意料之外的卡顿过后，就变成视频里的样子，我不能确定我在卡顿时完成的操作。
在Setting-Dimming settingsil-Dim lights to中的按钮可以调节这片区域的亮度。
我造成这一Bug时使用的是一个中文汉化版本，截图在附件中，但是我使用最新版也不能调整这片区域的灯效。
我尝试过清除注册表hkey_current_user\software路径下的alienfxfan、alienfx_sdk和alienfxgui文件夹，但是它不起作用。
重启电脑也没有作用。
我是否应该重置Bios？
这是我第一次使用GitHub，希望你能原谅我使用本软件和Github网站时的错误。
本篇由中文写作并使用机器翻译翻译成英文。
以下是中文原稿，您也可以使用不同的机器翻译它，以得到更好的译文。

TZSLLL, 2024-02-08:
This is a screenshot of the Chinese version
[image]

TZSLLL, 2024-02-21:
> "Devices and grids" tab is for define lights, not set it! First, set all lights from your system using it (it will be green one-by-one). THEN you finish, go colors/effects light to set it up to desired color/effect.

And I know that's not where the lights are set, but when I go to Settings, this is what it looks like
我也知道那不是设置灯光的地方，但是我转到设置之后特就是这样的

Uploading VID20240222050839.mp4…

TZSLLL, 2024-02-21:
https://github.com/T-Troll/alienfx-tools/assets/159496894/1ed030fd-369a-4303-85ec-30428a7d805e

TZSLLL, 2024-02-21:
https://github.com/T-Troll/alienfx-tools/assets/159496894/23e7edfb-2d39-49ab-ac1b-b22ea99f15cc

T-Troll, 2024-02-24:
Looks like you set your left zone as a hardware power button ("Power" checkbox at "devices" tab).
Please check it.

TZSLLL, 2024-02-24:
That's right, you helped me fix it!
So it's not a BUG, it's my mistake   
Anyway, thanks for helping me fix it.   :-)
wish you have a good day!

TZSLLL, 2024-02-24:
> Looks like you set your left zone as a hardware power button ("Power" checkbox at "devices" tab). Please check it.

That's right, you helped me fix it!
So it's not a BUG, it's my mistake
Anyway, thanks for helping me fix it. :-)
wish you have a good day!

closed 2024-02-24 by TZSLLL