T-Troll/alienfx-tools #412: Alienware M15R5 Gmode does not work :'(

EVilm1, 2024-02-18:
Hi, the problem is simple, I have an Alienware M15R5 (R9 5900X, RTX3070)
My system : Windows 11 23H2 / Tiny11 / AtlasOS

When I activate the Gmode from my keyboard (Fn+F1) it works fine (F1 key is white), it's managed by the bios. AWCC has never been installed and is totally disabled. The fans are set to maximum speed.

BUT, When I activate the Gmode from AlienFan-Gui or alienFan-CLI, my F1 key is white but nothing happens, the fans don't move and don't run at full speed.
Fans should be at full speed... but they're not.

Thx for your help.

EVilm1, 2024-02-20:
The difference is that when it's manual with the shortcut, it's like a "real game mode", with the fans at full speed.
I always thought it was the "Gamemode" but in reality it's the "level 153" mode that's activated.
When I activate the "Gamemode" via AlienFan, I can't see the difference with the "Manual" mode, the fans aren't running at full speed.

Btw I'm still looking for the best configuration for gaming, 100% Fan... (is there a risk for the fans to push them far at max boost?) And I'm trying to understand the CPU power AC mode, when it's in "enabled" mode my temperatures exceed 100° too quickly. when it's not I'm at 70° in game. I'm trying to find a middle point. The difference in game is huge about 30fps more...
Thx

T-Troll, 2024-02-24:
> "level 153" mode

Strange. It should be 0xAB, not 0x99.
Can you share me config (use `alienfx-config backup` command then send me `afx-fans.reg` file)?

EVilm1, 2024-02-24:
Of course! I didn't find anything weird in the .reg file.
[afx-fans.zip](https://github.com/T-Troll/alienfx-tools/files/14394664/afx-fans.zip)

T-Troll, 2024-02-25:
> is there a risk for the fans to push them far at max boost?

No one complain about it into the previous 2 years. We have one case of the broken fan, but it was a screw at the table (be careful then you put your notebook on it).

> And I'm trying to understand the CPU power AC mode, when it's in "enabled" mode my temperatures exceed 100° too quickly.

It's for AMD mostly, to have a little influence to Intel. Just switch internal CPU planning mode. In fact, it's just a hidden setting into the Windows Power Plan. You can google what is it about.

....

I check reg, it's ok. But i mention other - in fact, Fn+F1 handled by my GUI app, not BIOS! So the question is the difference between GUI and CLI. I will check how can i do it into CLI, it may need some fix.