T-Troll/alienfx-tools #415: 8.6.0 is getting blocked by Windows 11 as "Virus Detected" upon download.

dc-IV-AW, 2024-02-28:
I was able to use 8.6.0 as recent as today, but after a reboot, the Pinned Taskbar Icon would not launch. The .msi file has also been removed from my system. When attempting to download it again, it is blocked as "Virus Detected"

**To Reproduce**
Steps to reproduce the behavior:
1. Download 8.6.0 Release
2. Confirm Windows Defender blocks download as "Virus Detected."

**Expected behavior**
Latest .msi will download and be available in the local file system for installation.

**Screenshots**

[image]

[image]

**System (please complete the following information):**
 - Model: m16 R1 Intel RTX 4080
 - Model details: n/a
 - OS: Windows 11 Pro

**Additional context**
None

dc-IV-AW, 2024-02-28:
I checked in Windows Defender History:
[image]

dc-IV-AW, 2024-02-28:
> Reproduced on m15 R6, Windows 11 Home

Another strange issue is that older version's .msi is just fine on my system and not detected as a Virus:

Name: alienfx-tools.msi
Size: 905728 bytes (884 KiB)
SHA1: 5e73c42106afa39ad1068a454077ff9388726c0f

When I installed it, and checking "About" it shows the following:
[image]

So at least my Windows Installer folder still has the 8.6.0 version, and it is not getting flagged as a virus.

T-Troll, 2024-02-28:
It happened from time to time - being honest, i'm tired to fight AV's false positive.
I even have a page describing why this - https://github.com/T-Troll/alienfx-tools/wiki/Why-antivirus-complain-about-some-alienfx-tools-components%3F

100% of the source code available, no binary stubs, so you can clone repo, review and build yourself to be sure.

dc-IV-AW, 2024-02-28:
> It happened from time to time - being honest, i'm tired to fight AV's false positive. I even have a page describing why this - [...]

Maybe the issue should be a Feature Request to change **"In case you install tools version between 4.2.1 and 6.4.3.2, Anti-viruses can detect virus into project package."** 

to 

**"In case you install tools version between 4.2.1 and 6.4.3.2, in addition 8.6.0, Anti-viruses can detect virus into project package."**

T-Troll, 2024-02-28:
No, they always complain!
I drop kernel driver in v7 - this doesn't help!
The issue is i so close to hardware and the apps are unsigned.

Not only MS complain, other too. But others have a simple way to report false positive.

So... It's the question of trust. Build yourself, add to exception, do not use - it's all up to you.

dc-IV-AW, 2024-02-28:
OK, so being new to Alienxfx-tools, I did not realize it could still be flagged. I will close this since it is not something that can be fixed.

closed 2024-02-28 by dc-IV-AW