T-Troll/alienfx-tools #418: Uninstalling doesn't fully uninstall

LURSU-MiR, 2024-02-29:
I tried uninstalling and reinstalling the program, as there was a bug, I tried to fix, but the settings are still there even after a restart after the uninstall.

Steps to reproduce the behavior:
1. Uninstall Program
2. Restart
3. Installing the program
4. Settings are still there

My logical thinking is, that when I uninstall the program, that everything also the settings should be deleted.

 - Model: Dell G15 5515
 - OS: Windows 11

T-Troll, 2024-03-01:
Yes, it's true. Setting is not wiped at uninstall - in this case you will need to restore it any app update.
Instead, (as mentioned into tools readme - it's good idea to read it) i have `alienfx-config` too to manage configuration (backup/restore/delete). Check it.

LURSU-MiR, 2024-03-01:
how do I install alienfx-config?

T-Troll, 2024-03-01:
It's among the tools, installed into the same folder.

LURSU-MiR, 2024-03-01:
I found it but it instantly closed after opening

https://github.com/T-Troll/alienfx-tools/assets/161766591/9eafe55c-f01e-4b69-969d-41868d2f3541

T-Troll, 2024-03-01:
Uff....
1. Read project readme
2. Open CMD
3. Run it from it with parameters mentioned

LURSU-MiR, 2024-03-01:
Sorry for the unnecessary questions. I now got it deleted, but even now after a fresh reinstall it still happens that the fans just idle at 600 RPM no matter the temperature and only begin to spin again if I change the power mode to G mode, and switch back.
[image]

T-Troll, 2024-03-02:
It's a well-known bug of some hardware - fans stuck at low.
I fight it permanently, but it back again for new firmware....
Try to:
1. Set "Restore power mode" in settings.
2. Quit GUI
3. Set non-manual mode using `alienfan-cli`
4. Start GUI again

In this case, it will switch modes every time GUI/system start.

LURSU-MiR, 2024-03-02:
What is the command for the non manual mode

T-Troll, 2024-03-02:
0. Quit GUI, but set "Restore power mode" in settings before it.
1. Open **Administrator** CMD.
2. Navigate to the tools folder (`cd` command)
3. Run `alienfan-cli getpower`
4. You will see how much power modes your system have (at the first line) and current power mode next line (it will be `Manual (0)` most probably
5. Run `alienfan-cli setpower=X`, there X is something bigger, then 0 but lower, then total power modes you have
6. Look at the fans reaction (f.e. issuing `alienfx-cli rpm`)
7. In case it is OK, just keep it and run GUI

LURSU-MiR, 2024-03-02:
Thanks! Seems to be working now.

T-Troll, 2024-03-02:
Being honest, better downgrade BIOS as well - this bug can return in case mode will be still manual at system start.

LURSU-MiR, 2024-03-02:
Sure? Doesn’t the bios also have an effect on the security of the system?

T-Troll, 2024-03-02:
No one use flaws in "security" they patch.
The facts are you fans stuck in 0 boost then system start into Manual mode. And it's a bug.

Do as you wish, i've just provide information.

LURSU-MiR, 2024-03-02:
Ok I didn’t know that. if the problem reoccurs I will downgrade the bios.