T-Troll/alienfx-tools #426: Fans don't Spin according to the applied settings.

Bakr228, 2024-04-19:
Fans don't Spin according to the applied settings. 
I have an Alienware M18 R1 and it has a low fan speed cap at 1300RPM that starts the CPU1 and GPU3 fans at around 55C the Problem is the 1300RPM make a noise that no one can bare  so I had to use the AlienFX tool to fix it at first it didn't work when I had the 1.15.0 bios version because it always no matter what I Do in AlienFX both fans spins first at 1700 for 2sec and then to 1300rpm for 15 sec with the noise but when I took the risk to update the bios to 1.16.0 I became able to fix the CPU1 fan at 1500RPM when below 58C and not spin the GPU3 fan until it gets to 63C same about the other 2 fans the CPU2 and GPU4 
however today all this changed I can't fix the fans at 1500 it doesn't register sometimes it fixes at 2000rom sometime at 1800rpm even though I already did the check max boost and disabled AWCC.
Now both fans sync the start and fixes at 1300rpm and I have no idea how to fix it. 
do I need to do something in the advanced bios setting do I need to install the AlienFX again of what I'm Lost here Because I only Use AlienFx not The AWCC so it's either a fix or Myu PTSD Gonna break this Laptop when the Fans starts in quite mode.  

[image]

dc-IV-AW, 2024-05-13:
AFX Label | PCB Litho | Notes
---|---|---
CPU 1 | FAN1 | Shared Fan with GPU3/FAN2
CPU 2 | FAN3 | (MEMORY FAN, not really a CPU fan)
GPU 3 | FAN2 | Main GPU Fan shared with CPU1/FAN1
GPU 4 | FAN4 | Caddy Corner from FAN2, somewhat smaller with long vapor chamber

I don't know if this helps, but here is how the m16/m18 fans are represented in AlienFX.

T-Troll, 2024-05-23:
Fan control is indirect at our systems, so it can't be set below BIOS level, just above it (add more cooling).
So "boost", not a fan RPM percent, in fact, but how much power add to fan.

I got fan names from Alieware ACPI functions, so Dell threat it like this (and - by the way - only use one temperature to control it).