T-Troll/alienfx-tools #428: Dell G15 5525 RGB broken zone by AlienFX-cli

gjarro, 2024-05-22:
I was configuring the RGB zones on my G15 with AlienFX Control and it literally has locked the third zone of my RGB on a color I had set before. It does change the color but doesn't stay as it should, as it flickers back every some miliseconds to the color.
The steps to reproduce are literally trying to set up the lights using the color 180,240,60 on G15 5525, and I've seen a similar problem on a G15 5530 from my country, but in his case, it was the first zone of the keyboard, but same color.

I've attached a slow-mo GIF of the problem below:
[image]
[image]
[image]

System:
 - Model: Dell G15 5525
 - Model details: RTX 3060 with 4-zone RGB ABNT2 keyboard
 - OS: Windows 11 stable 23H2

ccrdmtr, 2024-05-22:
Hello, im the owner of the 5530 with the same problem, RTX 4050 model.

T-Troll, 2024-05-23:
Configs doesn't change hardware configuration into chip, as well as BIOS reset.
So, check some things:
1. You don't set this zone to "Pulse" at Color tab
2. You don't set this zone to "Power button" at Devices tab

In case this doesn't help - reset light chip - see project [readme](https://github.com/T-Troll/alienfx-tools?tab=readme-ov-file#light-system-reset) about how to do this.

gjarro, 2024-05-23:
I've downloaded an older version of the program, the 8.5, and tried messing up with some settings, and then it went back to normal, both on AlienFX and AWCC. It seems it could possibly be a bug from 8.6.

closed 2024-05-23 by gjarro