T-Troll/alienfx-tools #429: Fans don't spin using manual mode on G15 AMD

sa-zrx, 2024-06-02:
Every time I try to set my fans to spin according to the setting, they literally go to the lowest RPM possible and don't spin. Also, every time it tries to set the boost, my audio pops and there's a frame skip on my screen, as if it's trying to change the entire power mode.
I've tried to use manual mode on this laptop since I got it, and it had BIOS 1.8.0, and it's now on 1.16.1, none of the BIOS I've tested ever worked. Also, I've tried pretty much every version of AlienFX Tools since 8.3.0.

[image]
[image]

As the images below suggest, the boost dot alternates between red and green. Using the command "alienfan-cli.exe unlock" gets me exactly the same result. But, if I put a really high polling rate, the program is able to set the speed as it's intended to, yet it spins down to zero as soon as the time defined in ms passes and then goes up again depending on the temperature. For example, I've tried setting a different curve here:

[image]

Here's the system ID detected by the program:

[image]

I'm totally open to create any dump of this system if needed.
My system:
Dell G15 5525, (R7 6800H with RTX 3060)
Windows 11 23H2 stable
BIOS 1.16.1

sa-zrx, 2024-06-07:
I've tried downloading AlienFX 8.1.4 and somehow, the fan control works perfectly on this version. It would be very appreciated if this problem is investigated. Also, from 8.2.0 onwards, it absolutely stops working on my laptop.
[image]
[image]
Both GPU and CPU fans are working now accordingly to the curves and absolutely no audio pops of frame skips on my machine.

T-Troll, 2024-06-19:
Thanks. I'll check it then have a little time...

sa-zrx, 2024-07-05:
Issue fixed as of latest version (8.6.1.1)

closed 2024-07-05 by sa-zrx