T-Troll/alienfx-tools #433: How to reset all keyboard lighting settings

gjq722, 2024-06-29:
I accidentally exited  when I was setting up the lights for the first time, and now there is an area of ​​the keyboard light that is always orange. Now I can set the color of other area lights except that one. SO I want to reset all the settings related to the keyboard lights. I tried running alienfx-config.cmd in cmd, but it didn't work. What should I do,pls
[image]

gjq722, 2024-06-29:
Add:  In the device and grids tab, when I select this orange area, it flashes the highlight color and then turns back to orange, while the other areas always show the highlight color when selected.

sw0277, 2024-06-30:
Did you try to "light system reset" as explained here?

[https://github.com/T-Troll/alienfx-tools](url)

gjq722, 2024-06-30:
thank you ,The reply below this issue helpes me solve the problem ,have a good day  https://github.com/T-Troll/alienfx-tools/issues/428