T-Troll/alienfx-tools #434: Tool for linux

urbanze, 2024-07-08:
This tool works nicelly with Alienware m16 r1 (i9 13900HX + RTX 4060) in W11, but I use Ubuntu (24.04) most of time to to develop and work. Someone can give me any ideia to use this tool in linux?

I really want to control fans speed in linux (most important) and after this, of course, control some RGB LEDs to get more FPS on vscode xD. So, how fans are controled (BIOS, pwm, etc)? How LEDs are controled (bus, protocol, etc)?

T-Troll, 2024-07-09:
All you need to port my SDKs (Fan and FX). Now it uses Windows APIs, you need to alter it to Linux one. CLI will work after without any changes.
Most of the lights are USB HID devices, so it's simple.
Fans more tricky. SDKv2 utilize proprietary Dell WMI functions. Some Linux have WMI mappings as well, but it's Windows native in fact.
But SDKv1 uses direct ACPI methods call, so it's easy to port as well. The only issue, ACPI method names different from model to model, so you need to dump you BIOS and check.

t0m5k1, 2024-08-02:
I'm not a coder in any way so how do we go about doing any of this?

T-Troll, 2024-08-02:
I'm sorry, bro. Find a coder.

JL2210, 2024-08-25:
I'm interested in doing this. I think the first steps would be to switch build systems to CMake and then port all the USB HID code to a cross-platform library like hidapi.

The WMI calls are tricky. Some stuff is implemented in dell_wmi but it's really barebones. I'd probably have to write a bunch of stuff to get everything working.

 The ACPI calls should work. I might have to use the old `acpi_call` interface until a "proper" driver is written.

T-Troll, 2024-08-26:
Right.
However, i recommend to keep SDKs as library, so you can reuse it into different cli/gui.

ACPI have issue - main method name is quite stable (WMAX), but it's position into device tree and device name vary from system to system.
Good news - you can use mine [rwdec](https://github.com/T-Troll/rwdec) - it's WMI mappings decoder, it can help you to configure out whitch WMI device (AW**) has main method (AWCCWmiMethodFunction).

JL2210, 2024-08-26:
That's nice, thanks. I've been messing around with bmfdec but it seems to be missing the first 12 WMI methods from AWCC. Hopefully your tool can find them.

T-Troll, 2024-08-26:
NB: it's using .rw dump format, from RWEverything.
But it can find a lot, f.e. this is mine m16 dump.
[m16r1.txt](https://github.com/user-attachments/files/16743798/m16r1.txt)

urbanze, 2024-08-26:
I use linux but never developed something specific for linux (devices, drivers, etc). But Im embedded developer and have some skills to help this. Of course, I have one M16 R1 (i9 13900HX + 4060) and can help with tests and so on.

JL2210, 2024-08-26:
I think the best way for me to work on this is to dual-boot both Windows and Linux. Wine and QEMU device pass-through don't work super well with this.

ACPI calls on Linux are somewhat hard to do well. You either need a specific driver to abstract the calls to something more user friendly or you can use `acpi_call` which creates a file at `/proc/acpi/call`.

In my experience `acpi_call` is kind of fiddly and hard to get working well (read: you need to close and re-open the file every time you want to make an ACPI call), so I'll eventually want to put something in the `alienware-wmi` driver.

T-Troll, 2024-08-26:
Ok, guys, i wish you good luck and feel free to ask me about details.
F.e., HID interfaces uses all 3 types of communications - Report, Feature_set, Interrupt. You should support it all.

urbanze, 2024-12-19:
[acpi_tables.txt](https://github.com/user-attachments/files/18202059/acpi_tables.txt)

I get acpidump from my alienware m16 r1 (i9 13900Hx + rtx 4060) but I cant find any path do test some fan commands. Im really lost about everything, but will try with some help

T-Troll, 2024-12-23:
I have a small tool here - [rwdec](https://github.com/T-Troll/rwdec).
You can use it to configure out which ACPI method exposed to Windows as AWCC control block. Then use it.

kuu-rt, 2024-12-25:
Hello!

Support for switching thermal profiles on Linux has already been merged upstream and will be available on Linux v6.13. To test it before it hits stable, pass force_platform_profile=1 parameter to the alienware_wmi module. 

Your model supports the WMI interface, so I will submit a patch adding your model to the allowlist, so you won't need the module parameter.

You can switch thermal profiles using the power slider on KDE and Gnome or by using the platform profile interface, which has more profile choices [1].

If you manage to successfully test it, I can add a Tested-by tag when submitting the patch.

[1] https://docs.kernel.org/admin-guide/abi-testing.html#abi-sys-firmware-acpi-platform-profile

T-Troll, 2024-12-25:
Wow! Hope they configured out it just need to check AWCC WMI function exist to be sure it's Alienware/G-Series. And also how to control fans.
Anyway, good news!

kuu-rt, 2024-12-25:
> Wow! Hope they configured out it just need to check AWCC WMI function exist to be sure it's Alienware/G-Series. And also how to control fans. Anyway, good news!

Unfortunately there is not a way of decoding bmof data in kernel-space, so every supported supported model has to be added to an allowlist for now. That's why I implemented force_platform_profile and force_gmode kernel parameters to let users bypass this allowlist and tests this new features.

I will submit patches for manual fan control with the custom profile (0x00) through the HWMON interface soon. Another feature which was recently merged was needed before that.

I am the author of these new Linux features so if anyone has questions of suggestions, let me know!

urbanze, 2024-12-25:
@kuu-rt Amazing! Sorry for my lack of in-depth knowledge about Linux.

How can I test this on Ubuntu before waiting for canonical releases a new version with this kernel? Or I will need to wait? I really want this features and will use daily.

urbanze, 2024-12-25:
Oh, I installed v6.13-rc1 sucessfully on my ubuntu 24.04! I trying to use this force_platform_profile=1 and finding where/how to change profiles

kuu-rt, 2024-12-25:
@urbanze I'm glad it worked :)

Is it ok if I use your name and email when submitting the patch adding your model to the allowlist? 

> @kuu-rt Ok, this is working at m16 r1. How can I set fan threshold like AWCC, for example to stay fan at least 30% (offset)?

Manually controlling fans is not supported yet, but I'm already working on it!

urbanze, 2024-12-26:
> Is it ok if I use your name and email when submitting the patch adding your model to the allowlist?

@kuu-rt I don't see any problems or consequences, is there anything I should know if I use it? If there is nothing, you can use it. If the community is going to use it to email me with this feature or bugs, then it's better not to use it because I have no idea how to do it.

Please let me know when you implement this feature and I will test it the same day! This will help me a lot in keeping my laptop cool so I can use it on my lap from time to time.

T-Troll, 2024-12-28:
> I am the author of these new Linux features so if anyone has questions of suggestions, let me know!

Oh, i glad to hear someone start to do real stuff!

> Unfortunately there is not a way of decoding bmof data in kernel-space

It's not so hard, check mine [rwdec](https://github.com/T-Troll/rwdec) tool - you need to find mapping for AWCCWmiMethodFunction. You don't need to do it in kernel, by the way, but just map once into settings.
I'm asking people for BIOS dump before then just analyze it and then add new model.

Also, check mine Fan SDK v1 (direct ACPI) - it have some hints about how to detect and use DPTP/KRDT temperature sensors present into hardware. Quite easy too.

Also, i attach mine working notes here - it's about call mappings for ACPI main function (it's always the same for any notebook, but different for desktops - shifted by -4). And some rwdec result for different models - it have a hint for function codes.

[info.zip](https://github.com/user-attachments/files/18265536/info.zip)

kuu-rt, 2024-12-30:
> > Is it ok if I use your name and email when submitting the patch adding your model to the allowlist?
> 
> @kuu-rt I wouldn't see any problems, but since someone might get in touch claiming bugs or problems, thinking that I can solve them, it's better for me to stay out of it, since I won't be able to help at all LOL

@urbanze  Oh don't worry. It's just a way to give credit to testers on the commit message, you are not expected to solve anything :)

> Please let me know when you implement this feature and I will test it the same day! This will help me a lot in keeping my laptop cool so I can use it on my lap from time to time.

For sure! It will be finished and submitted upstream in the following weeks.

kuu-rt, 2024-12-30:
> Oh, i glad to hear someone start to do real stuff!
>
>
> It's not so hard, check mine [rwdec](https://github.com/T-Troll/rwdec) tool - you need to find mapping for AWCCWmiMethodFunction. You don't need [...]

In Linux there is no easy way to call WMI (or any ACPI method for that matter) in userspace without custom drivers. That's why the alienware-wmi Linux driver needs to know AWCCWmiMethodFunction mapping ahead of time, which means I have to manually specify each model's quirks.

Of course that's very tedious, but unfortunately there is no other way for now.

> 
> Also, check mine Fan SDK v1 (direct ACPI) - it have some hints about how to detect and use DPTP/KRDT temperature sensors present into hardware. Quite easy too.

I'll check it out!

> Also, i attach mine working notes here - it's about call mappings for ACPI main function (it's always the same for any notebook, but different for desktops - shifted by -4). And some rwdec result [...]

This and Fan SDK v2 are helping me a lot in understanding some of WMAX's methods. 

Thank you so much @T-Troll. I'll be sure to credit you once I submit the manual fan control and sensor monitoring features upstream :)

T-Troll, 2024-12-30:
> In Linux there is no easy way to call WMI (or any ACPI method for that matter) in userspace without custom drivers. That's why the alienware-wmi Linux driver needs to know AWCCWmiMethodFunction mapping ahead of time, which means I have to manually specify each model's quirks.

You don't need WMI, in fact. You just need to find all ACPI devices with _HID = PNP0C14 and then decode WQMO buffer to configure out is WMAX on this device mapped to AWCCWmiMethodFunction or not.

tr1xem, 2025-11-01:
Hello Im working on [AWCC](https://github.com/tr1xem/AWCC) for linux.
This project have been a great resources
I had some doubts about somethings

- currently i have seen their are mostly 2 prefix which are `AWM3`(for amd),`AMWW` (for intel) but oddly some amd device need intel prefix and stuff like aurora need `AMW1` (decompiled its acpidump) although it have same interface but it got some undocumented modes (which i found though looking at acpidump again)
```
First one:
\_SB.AMW1.WMAX 0 0x15 {0x01, 0x197, 0x0, 0x00}
0x0
After first command: _SB.AMW1.WMAX 0 0x14 {0x0b, 0x00, 0x00, 0x00}
0x97⏎

Second one:
\_SB.AMW1.WMAX 0 0x15 {0x01, 0x198, 0x0, 0x00}
0x0⏎    
After second command: _SB.AMW1.WMAX 0 0x14 {0x0b, 0x00, 0x00, 0x00}
0x98⏎
```

Which could be why dell makes 2 app for windows (my guess)?
In your discovery do they have more prefix? if yes then do they have diff methods?

- Regarding lights , my current approach is trying to capture packets from various device and then reverse and copy them using libusb which have been working for most of the part (only per key rgb is remaining now)

how do u find zone code for per key rgb? from the dll i decompiled from offical awcc they could send more than 1 buffer for all keys

- About the custom mode i dont see any difference in custom mode? i can just set cpu gpu boost in any mode why use custom mode?

@kuu-rt ur feedback would be quite usefull

T-Troll, 2025-11-01:
> In your discovery do they have more prefix? if yes then do they have diff methods?

Yes, it's model-depended. For clear detection, you can use mine [rwdec](https://github.com/T-Troll/rwdec) to find where AWCC method mapped to.

> Regarding lights, my current approach is trying to capture packets from various devices and then reverse and copy them using libusb, which has been working for most part (only per-key RGB is remaining now)

You can port mine SDK (just change device detection from Windows to Libusb). In this case, we can share new API and mods with easy.

> How do u find the zone code for per-key RGB? From the DLL I decompiled from the official AWCC, they could send more than 1 buffer for all keys.

AWCC doesn't have it - they download map online using service tag as a key.

> About the custom mode, I don't see any difference in the custom mode. I can just set CPU and GPU boost in any mode. Why use custom mode?

Some hardware can do this trick, but most just doesn't allow setting, or it ignored.