T-Troll/alienfx-tools #439: alienfx-config.cmd

TheSQLGuru, 2024-08-05:
1) Win 11, M18 R2, AFX version 8.6.1, alienfx-config.cmd date 202210180659

2) I ran alienfx-config.cmd:

`D:\Program Files\Utilities\AlienFXTools>alienfx-config.cmd backup
Usage: alienfx-config [command]
Commands are:
        backup  - save registry config to files. 4 config file will be created into current folder.
        restore - load config files back to registry.
        delete  - remove all tools configuration from registry.
Doing backup...
The operation completed successfully.
The operation completed successfully.
ERROR: The system was unable to find the specified registry key or value.
ERROR: The system was unable to find the specified registry key or value.
Done.` 

3) In my D:\Program Files\Utilities\AlienFXTools directory, I have these two new files:

afx-fans.reg
afx-lights.reg

Contents of afx-fans.reg:

`Windows Registry Editor Version 5.00

[HKEY_CURRENT_USER\Software\Alienfan]
"StartAtBoot"=dword:00000001
"StartMinimized"=dword:00000001
"LastPowerStage"=dword:00000000
"UpdateCheck"=dword:00000000
"DisableAWCC"=dword:00000000
"KeyboardShortcut"=dword:00000000
"KeepSystemMode"=dword:00000001
"DPTF"=dword:00000000
"PollingRate"=dword:00000535
"Boost-0"=hex:65,69,10
"Boost-1"=hex:65,67,23
"Boost-2"=hex:65,c5,0f
"Boost-3"=hex:65,4e,1a

[HKEY_CURRENT_USER\Software\Alienfan\Powers]
"Power-0"="Manual"
"Power-160"="Level 160"
"Power-161"="Level 161"
"Power-162"="Level 162"
"Power-163"="Level 163"
"Power-164"="Level 164"

[HKEY_CURRENT_USER\Software\Alienfan\Sensors]
"Fan-0-257"=hex:00,00,23,00,23,05,29,05,29,0a,33,0a,41,28,4b,64,64,64
"Fan-1-257"=hex:00,00,23,00,23,05,29,05,29,0a,33,0a,41,28,4b,64,64,64
"Fan-2-262"=hex:00,00,23,00,23,05,2a,05,2a,0a,33,0a,3c,28,46,64,64,64
"Fan-3-262"=hex:00,00,23,00,23,05,2a,05,2a,0a,33,0a,3c,28,46,64,64,64
"SensorName-0-0"="ESIF sensor 1"
"SensorName-0-1"="ESIF sensor 2"
"SensorName-0-2"="ESIF sensor 3"
"SensorName-0-3"="ESIF sensor 4"
"SensorName-0-4"="ESIF sensor 5"
"SensorName-0-5"="ESIF sensor 6"
"SensorName-0-6"="ESIF sensor 7"
"SensorName-0-7"="ESIF sensor 8"
"SensorName-0-8"="ESIF sensor 9"
"SensorName-1-1"="CPU Internal Thermistor"
"SensorName-1-6"="GPU Internal Thermistor"

`

Contents of afx-lights.reg (no configuration for these):

`Windows Registry Editor Version 5.00

[HKEY_CURRENT_USER\Software\Alienfx_SDK]
`

4) If these are the only two files I should have given my configuation, then eliminate the two error messages. If not, I will be happy to help discern the issue. 

5) I will update to the just released 8.6.1.2. As expected, I see that the alienfx-config.cmd file did not change date. So no need to retest.

T-Troll, 2024-08-05:
1. It should be 4 files, but seems like you never run FX and Mon. So errors are OK in this case.
2. Interesting. EMPTY SDK config. So you don't define lights.
3. cmd is a simple script to dump/remove/load registry configuration. So it not changes most of the time.

The only suspicious is empty SDK record. The rest are ok.

TheSQLGuru, 2024-08-05:
Error messages to the end user are not OK if they are not actually errors. :-) I did not thing wrong, and it is not an error to have not ever run those two apps. Simply check for the condition and skip extract if not applicable. And make sure the user knows that there should only be as many files as used applications. Poof, done. ;-P

Correct, I have not used lighting on this laptop since to my knowledge my M18 R2 is not supported. And I have not had a chance to collect the information for you.

P.S. Stupid github page won't let me remove the emoji I put on your response by mistake.