T-Troll/alienfx-tools #441: Aurora R13 "No light defined"

crsccc, 2024-08-16:
I purchased a Solid Side Panel version of the R13, and i want to upgrade to the Clear Side Panel version,  I’ve purchased the RGB components that are standard on the Clear Side Panel version. However, AWCC cannot recognized these new components, so I install AlienFX tool, but it always shows "no light defined", and there is no light zone show up in the tool.

**System (please complete the following information):**
 - Model: Alienware Aurora R13
 - OS:  Windows 11

T-Troll, 2024-08-20:
Seems like you didn't connect it correctly - so USB device (light chip) doesn't recognize into the system.
Hint: You should have USB HID device with the VID_187C, PID_0550 among other USB HIDs.

crsccc, 2024-08-20:
> Seems like you didn't connect it correctly - so USB device (light chip) doesn't recognize into the system.
> Hint: You should have USB HID device with the VID_187C, PID_0550 among other USB HIDs.

thanks for your reply. i do not quite understand that the front panel light and cooling light are directly connected to the motherboard.

T-Troll, 2024-08-20:
I don't sure about direct connection, but it should be connected to MB or other extension board (i don't know how R8 connected).

crsccc, 2024-08-20:
here is my motherboard https://dl.dell.com/content/manual40490763-alienware-aurora-r13-service-manual.pdf?language=en-us

the front ambient light connects to #1, and the cooling rgb light connects to #20. I also have 3 fans connected, but none of them shows on the FX tools
> I don't sure about direct connection, but it should be connected to MB or other extension board (i don't know how R8 connected).
> JFYI: https://m.media-amazon.com/images/I/B1iBHYQ58pS.pdf (service manual) - check connector #17