T-Troll/alienfx-tools #444: Add support for Area 51 R4

grghub, 2024-08-28:
I have an Alienware Area 51 R4, which is listed as "Unknown" Tron in the "Supported and tested devices list", and hope I can provide enough info here to allow it to be known and supported.

In alienfx-gui the "Detect devices" button didn't find anything, but `alienfx-cli probe=a` did actually find all the lights (although I don't have it fully working, see below) -- so I'll start here with what that found in case that's an easy source of accurate info.

```
C:\Program Files\AlienFX Tools>alienfx-cli.exe status
alienfx-cli v8.6.1.1
Dell API not found, 1 devices found.
Device #0 - Alienware Area 51  R4, VID#6268, PID#1318, APIv2, 9 lights
  Light ID#0 - right panel, top
  Light ID#1 - right panel, forward
  Light ID#10 - alien head power button (Power button)
  Light ID#11 - front panel, left
  Light ID#12 - front panel, right
  Light ID#2 - right panel, rear
  Light ID#5 - left panel, top
  Light ID#6 - left panel, rear
  Light ID#7 - left panel, forward
1 groups:
  Group #0 (9 lights) - all lights
Done.
```

An `alienfx-config.cmd backup` produces the following afx-lights.reg file (renamed to make github allow the upload): [afx-lights-dotreg.txt](https://github.com/user-attachments/files/16786271/afx-lights-dotreg.txt)
And alienfx-gui's "Save Lights and Grids" produces [Current.csv](https://github.com/user-attachments/files/16786605/Current.csv)

With that configuration, `alienfx-cli.exe setall=255,0,0` does turn all case lights except the power button red, and `alienfx-cli.exe setone=0,10,0,255,0` turns the power button green.

However, I couldn't get the `setpower` command to work; variations of `alienfx-cli.exe lowlevel setpower=0,10,255,0,0,0,0,255` all seemed to hang for a minute or so and then exit without doing anything.

Furthermore, I tried to make alienfx-gui do interesting things with this setup and started by trying to enter a "Grid Effect" with one zone of all 9 lights, Trigger=Continuous, and when nothing happened I added a few Colors and tried various checkboxes on that screen, but then alienfx-gui started crashing.  Upon restart it would hang for a minute, work for a few seconds, then hang for another minute and crash.  Across several restarts during those few working seconds I manage to remove all the colors and uncheck all the checkboxes, and once I got that done and Saved it stopped crashing (but of course does nothing, given no instructions).  Maybe this is just my ignorance and I'm doing the wrong thing?  I admit that I have no idea how to tell that gui to do something interesting.

Following the "How to collect data for the new light device" instructions -- from the info above I think the relevant device is 
HID\VID_187C&PID_0526 (which is also listed in https://github.com/trackmastersteve/alienfx/blob/master/alienfx/data/etc/udev/rules.d/10-alienfx.rules as the device for the Area 51 R2 (the R4 is not listed there, but it looks like maybe it's the same as the R2?); here are the all the Hardware IDs listed for that device in Device Manager:
```
HID\VID_187C&PID_0526&REV_0000
HID\VID_187C&PID_0526
HID\VID_187C&UP:FF00_U:0001
HID_DEVICE_UP:FF00_U:0001
HID_DEVICE_UPR:FF00-FFFF
HID_DEVICE
```

I think that's what you're looking for, but for completeness I'll mention that there's a standalone top-level "Alienware Alienware Area-51 R4" device with hardware IDs:
```
COMPUTER\{CBEE6E87-CA56-5645-A998-11643185BCD0}
COMPUTER\{140FCCD7-1041-548C-9222-7E22B756E90B}
COMPUTER\{6C0351A2-983A-58AA-A36D-153489C028C9}
COMPUTER\{ABACDA57-53F5-506D-B968-E4A9CD1ABDDE}
COMPUTER\{32F32D84-A1D0-5D3A-A2ED-B8E9ABD2077A}
COMPUTER\{F74D7841-CAA7-5276-87F1-4C59683694E4}
COMPUTER\{BE2EC741-44E6-5A68-BAE3-E99F5739831D}
COMPUTER\{C2AF4C2E-274F-5D8E-A913-0F874C798E90}
COMPUTER\{37AECE23-4ECF-5136-ADD2-0E021EE4E690}
COMPUTER\{73735569-B868-5294-B874-2E0D995F1F02}
COMPUTER\{57C07333-E98C-5D18-BDC1-C1B8F6E7491F}
COMPUTER\{1DB2BEFB-DC91-5482-8BF7-3F65C31BB628}
and "Bus Type GUID" property = {06d10322-7de0-4cef-8e25-197d0e7442e2}
```

and also a second "HID-Compliant Vendor-Defined Device" with hardware IDs:
```
HID\VID_0B0E&PID_0412&REV_0109&MI_03&Col02
HID\VID_0B0E&PID_0412&MI_03&Col02
HID\VID_0B0E&UP:FF00_U:0001
HID_DEVICE_UP:FF00_U:0001
HID_DEVICE_UPR:FF00-FFFF
HID_DEVICE
```

I don't have AWCC running, but here is a USBPcap of making light#0 (right panel, top light) green (0,255,0): [caselightgreen-pcap.log](https://github.com/user-attachments/files/16787008/caselightgreen-pcap.log)

The instructions ask for dimming and other effects, but since I don't have AWCC or alienfx-gui running, the only way I could think of to get any kind of effect recorded was to capture while sleeping & waking the machine (sleep = turns off all case lights, makes alien head power button fade off & on orange; and wake = turns on all case lights to aqua color, makes alien head power button green), with the caveat that I don't know which of those light actions were actually captured in the pcap vs which were done when the capture process was asleep (but the file looks like the right size to have captured something?): [sleep-pcap.log](https://github.com/user-attachments/files/16787038/sleep-pcap.log)

Hope the above contains the info you're looking for, but of course I'd be happy to provide any other information I can find, run more captures, or other tests as needed.  (And as an aside, if I'm doing something wrong in alienfx-gui, I'd appreciate knowing what to tell it to get it to do something interesting.)  Thank you!

T-Troll, 2024-08-29:
Well... Impressive job, thank you!
I'll add your mappings into autodetect next release.

Now about issues.... For APIv1,v2,v4 light system can spend a lot of time then you try to set power button.
But app should not hang, i use separated thread for it.
BTW, making light as a HARDWARE power button enable automation - so charging/sleeping/battery will be automatic as you set it.

So if app hangs, issue is out of it, thanks for pointing this, i'll take a look.

The bad news is my last APIv2 device was broken beyond repair 2 years ago, so i can't check what wrong with power button code directly.... I only can guess (it works before, but i can break it).

> and when nothing happened

Grid effects REQUIRE lights located on grid. I check about crash, most probably it's in case some of them misplaced. You are share me enough information for this. Also, please share me afx-gui.reg to understand you effect settings.

grghub, 2024-08-29:
Here's afx-gui.reg as it stands now, which is after all the colors were removed & checkboxes cleared so it doesn't crash (filename-mangled for github): [afx-gui-dotreg.txt](https://github.com/user-attachments/files/16796653/afx-gui-dotreg.txt) -- I think it should show that I have one zone with all 9 lights, but absolutely nothing else set... (although looking at the file I don't see info on the zone? curious..)

Current.csv above does I think show that I have the power button marked as a hardware power button, FWIW.

Does "grid" mean only keyboard lights?  (which I don't have)  Sorry, I'm new to alienfx-tools and still figuring it out...

I think I should clarify that I do have the latest AWCC installed, but it doesn't seem to support this hardware (while the hardware clearly works, since alienfx-tools successfully controls all 9 lights) - could that be interfering?  Should I uninstall AWCC or otherwise disable it somehow?

Thank you!

T-Troll, 2024-08-29:
> Here's afx-gui.reg as it stands now, which is after all the colors were removed & checkboxes cleared so it doesn't crash (filename-mangled for github): [...]

Oh.... I need one then it crashes/hangs...
And yes, zones into lights.reg

> Does "grid" mean only keyboard lights? (which I don't have) Sorry, I'm new to alienfx-tools and still figuring it out...

No. Grid means ANY light position at desktop/notebook surface. GRID effects only available if light position defined on grid.
You can make grid for any side of notebook, or (additionally) grid aggregating all lights.
NB: Light can be placed to more, than one grid.

> I think I should clarify that I do have the latest AWCC installed

No, it doesn't matter. You can have issues with supported lights, mostly, and fan control.

BTW.... Fan control. As i can see, you have it enabled. Did it work for your gear? Should not, this control standard come later, then any v2 device... THIS can be the source of issues.

grghub, 2024-08-29:
> Oh.... I need one then it crashes/hangs...

OK, I experimented some to figure out the crash/hang of alienfx-gui (recall from above that `alienfx-cli setpower` also hangs, but that's not covered here).  alienfx-gui will get into its crashing mode if it's configured with all of these three things:
1. Trigger = Continues (on Grid Effect tab)
2. colors in the Color List (on Grid Effect tab)
3. a zone defined with lights **which includes the power button** (yes, I know you said to take it out; here I'm just debugging the hang, and your response about it is what called my attention to this)

If those three things are configured (here they are in an afx-gui.reg: [afx-gui-all9-crashes-dotreg.txt](https://github.com/user-attachments/files/16801283/afx-gui-all9-crashes-dotreg.txt)), if I start alienfx-gui and just leave it, it crashes about 55 seconds later.  It does seem to be operable during that time, without hanging.  The hang happens if I click on the "Devices and Grids" tab - if I do, it hangs for about 40 seconds before filling out the display for that tab.  (And if it hits the 55-second mark before those 40 seconds are up, it crashes in the middle of hanging.)

If I take any of those 3 away (delete the colors, or make Trigger=off, or take the power button out of the zone leaving the remaining 8 lights) then alienfx-gui doesn't crash.  (And taking the power button out of the zone, it "only" takes about 20 seconds of hanging before displaying the "Devices and Grids" tab.)

> BTW.... Fan control. As i can see, you have it enabled. Did it work for your gear?

I didn't enable fan control -- maybe it's enabled by default?  I haven't done anything with the fans.  Reading the docs, it looks like the way to disable fan control is to change "Power mode" to anything other than "Manual"?  But the only option alienfx-gui  gives me for Power mode is Manual, I can't change it to anything else.

Finally, I still have the issue which started after running the above alienfx-cli commands, which is that when I power the machine on (without running alienfx-tools), the 3 lights on the right panel do not come on, whereas they always had before (but again, the other 6 case lights do come on).  Any idea what I need to do to get it back to turning on all the lights at boot?  Thanks!

T-Troll, 2024-08-30:
About crashes - seems like i found the reason, will fix into new release.

About hangs... The source is understandable. Issue is you chip report non-ready state, so setting each light wait for ready until timeout (about 4 sec per light). Devices&Grids do not use separate thread for fading lights, so... yes, about 35 sec hang.
Seems like i make a mess with a status check then modify SDK... But i doesn't have v2 now for check and fix... Ok, let me think.

About fan - just turn it off in Settings. Seems like your BIOS have functions needed for it (so support detected), but they are not implemented.

CLI does not save colors, it can just change it until reboot. Use GUI "Save" button to store current light set.

grghub, 2024-08-30:
I have some color cycling mostly running now!  [afx-lights-dotreg.txt](https://github.com/user-attachments/files/16821063/afx-lights-dotreg.txt), [afx-gui-dotreg.txt](https://github.com/user-attachments/files/16821062/afx-gui-dotreg.txt)

But one apparent issue here -- with it running from the "Colors" tab (several colors, morphing), on its first run-through when it switches to 2nd & 3rd colors in the list, light #7 blinks off and then on; and then on 4th color in the list it just goes off & stays off until the cycle is restarted (e.g. by changing something in the config).  (Is this somehow related to not having full support for this device yet, or should I be filing this as a separate issue?  I don't want to abuse and muddle up this one issue...)

> Please REMOVE power button from "all lights" zone and set it separately, then needed (as i told you, it's a great slowdown).

I'm not sure if this is what you were saying, but if I have the power button in **any** zone, even if that zone isn't on the grid, alienfx-gui crashes.  (Is this related to the crash you already found (thanks!!), or a different one?)

> CLI does not save colors, it can just change it until reboot. Use GUI "Save" button to store current light set.

Using the GUI Save button does not seem to affect what happens on cold boot, so I still now have a dark right panel at power-on.  It seems like that would have to be a BIOS/CMOS setting?  The lights come on as soon as the power comes on, seemingly before even a POST.  and somehow that's now been changed, with 3 of the lights (all on the right panel) not lighting by default any more (the rest of the lights come on as they always have).  Anything else I can try to get that restored?

EDIT: Oh, wow, after playing around with alienfx-gui some more I'm in a state now where ONLY light#7 comes on at cold boot, with one of the colors I was using in the GUI, the rest are all dark.  Clearly something I'm doing is affecting this -- what?  How do I fix it back?

> About fan - just turn it off in Settings.

Done, thanks for the tip.  FWIW, when it was on it seemed to be displaying actual data for 3 fan speeds and 9 temperatures.  but it didn't show any fan speed curves and I didn't try controlling the fan speeds.

Thank you!

T-Troll, 2024-09-01:
> Is this somehow related to not having full support for this device yet, or should I be filing this as a separate issue?

it's just because your device only support 2-color effects. My app trying to extend it, but reaction can be... strange.

> I'm not sure if this is what you were saying, but if I have the power button in any zone, even if that zone isn't on the grid, alienfx-gui crashes. (Is this related to the crash you already found (thanks!!), or a different one?)

Wait a moment. We don't talk about grid, but set. What if PB is in a separate zone, but not set to anything? What if set TWO colors?

> The lights come on as soon as the power comes on, seemingly before even a POST. and somehow that's now been changed, with 3 of the lights (all on the right panel) not lighting by default any more (the rest of the lights come on as they always have). Anything else I can try to get that restored?

Your device have a special commands to save light colors for system start. Seems like this lights not saved correctly. This can be related to 4-color set attempt.

> Done, thanks for the tip. FWIW, when it was on it seemed to be displaying actual data for 3 fan speeds and 9 temperatures. but it didn't show any fan speed curves and I didn't try controlling the fan speeds.

You can play with it if you wish, try to define some curves and see what happened. I just not sure if it will work for your system (but should not break anything).

grghub, 2024-09-01:
I see the release note for v9.0.0.0 says it has Area-51 R4 profile detection in the gui!  Thank you for the speedy support!  Weirdly, the first few times I tried the "Detect devices" button on the v9.0.0.0 gui (after clicking on "Clear device" to get a cleaner slate) it didn't seem to detect anything, but after playing around a bunch more (including the below `alienfx-cli probe=a`? and a bunch of reboots) now it does detect the Area-51 R4 and load the light list!

> Wait a moment. We don't talk about grid, but set. What if PB is in a separate zone, but not set to anything? What if set TWO colors?

I experimented more with this, here's what I found:
* the PB can be in a zone as long as that zone has no colors set and it won't crash
* if the PB zone has colors set the gui crashes.  I tried 1 and 2 colors, and color modes Color, Pulse, and Morph; all crash.
* If there's a saved configuration with colors set on the PB zone when the gui starts up, the gui crashes 56 seconds after startup.  It crashes even if during those 56 seconds I remove the colors and save the configuration.
* if I add a color to the PB zone (which on gui start had no colors), the gui crashes 56 seconds after I add the color.  Specifically, it's 56 seconds after I click on the color in the color picker (doesn't matter how much time elapses after I click the "+" button to start adding the color, and I don't even have to click the OK button on the color picker for it to crash)

> it's just because your device only support 2-color effects.

Oh!  I guess I shouldn't be trying more than 2 colors per zone and expecting it to work on cold boot then?  I had no idea.

(Did I miss that in the docs?  I saw in the doc page Breath, Spectrum, and Rainbow all said they needed APIv4, which is why I was using Morph.  And I did see that it says Morph has a *minimum* of 2 lights.  But at the bottom of the `alienfx-cli` help output under all of the actions including Morph it says "Up to 9 colors can be entered", so I was trying even up to 8!  And 8 "mostly" worked in the gui (except light #&#x2060;7 blinking and then staying dark), but not on power-up.  Maybe a warning message in the gui would be helpful?)

### alienfx-cli

Even with only 2 colors for "all but PB" zone and no colors for PB zone, `alienfx-cli` seems to be having some problems now; last week `alienfx-cli setone` and `alienfx-cli setall` always worked, but today they almost always fail to do anything when run... I've seen `alienfx-cli setall` work only once in the middle of a long series of attempts, and I haven't seen `alienfx-cli setone` work even once today.  I ran through an `alienfx-cli probe=a` again - which does turn all lights off and then each one on at a time, so alienfx-cli does still seem to have some light control (and I hoped that running through the probe might reset something to a working state?  but `setone` and `setall` still weren't working even after the probe).

At first I was worried it might be a regression added in v9.0.0.0, but then I noticed the alienfx-cli output reports it's v8.6.1.1 (as it did when I had the v8.6.1.3 msi installed), which in itself seems odd...

T-Troll, 2024-09-13:
I'm sorry for later reply - i just have my main duty calls me sometimes.
So...

What i need to mention - Power button control for V2 thrashed into latest releases.
Also, i need to mention i doesn't have APIv2 device for check and fix.

So let's go the other way. Try to install older v8.x.x.x versions and check power button set (you can use CLI).
In case we can pinpoint what version broke the stuff - i can fix it back.

grghub, 2024-09-17:
### Power Button

> So let's go the other way. Try to install older v8.x.x.x versions and check power button set (you can use CLI).

OK, I have those results for you, but it's not a simple story - since v8.0.0.0, whether or not the PB can be controlled via `alienfx-cli setone` has changed at least 5 times:
* it worked up through 8.0.2
* it stopped working in 8.1.0
* it worked again in 8.1.0.3, but only that single version
* stopped working again in 8.1.1
* started working again in 8.2.6.2 (and ran noticeably faster!), through 8.5.2.2
* stopped working again in 8.6.0 (and hasn't worked since)

For completeness, here's the list of versions I tried (which doesn't cover every 8.x.x.x version, so there might be some short-lived changes like 8.1.0.3 that I've missed?), whether or not `alienfx-cli setone=0,10,0,255,0` on the PB light controlled it, and how long the command took to run:
 * **YES** 8.0.0 - 11 sec
 * **YES** 8.0.1 - 11 sec
 * **YES** 8.0.2 - 11 sec
 * NO  8.1.0.0 - 3 sec
 * NO  8.1.0.1 - 3 sec
 * NO  8.1.0.2 - 3 sec
 * **YES** 8.1.0.3 - 11 sec
 * NO  8.1.1 - 0.2 sec
 * NO  8.1.2 - 4 sec
 * NO  8.1.3 - 4 sec
 * NO  8.1.4 - 4 sec
 * NO  8.2.0 - 4 sec
 * NO  8.2.5 - 4 sec
 * NO  8.2.6.0 - 4 sec
 * NO  8.2.6.1 - 4 sec
 * **YES** 8.2.6.2 - 2 sec
 * **YES** 8.2.6.3 - 2 sec
 * **YES** 8.2.6.4 - 2 sec
 * **YES** 8.2.6.5 - 2 sec
 * **YES** 8.2.6.6 - 2 sec
 * **YES** 8.2.6.7 - 2 sec
 * **YES** 8.2.7 - 2 sec
 * **YES** 8.3.0 - 2 sec
 * **YES** 8.4.0 - 2 sec
 * **YES** 8.4.1 - 2 sec
 * **YES** 8.5.0 - 2 sec
 * **YES** 8.5.2.0 - 2 sec
 * **YES** 8.5.2.1 - 2 sec
 * **YES** 8.5.2.2 - 2 sec
 * NO  8.6.0.0 - 2 sec
 * NO  8.6.1.0 - 2 sec
 * NO  8.6.1.3 - 2 sec
 * NO  9.0.0.0 - 2 sec
 * NO  9.0.0.1 - 2 sec
 * NO  9.0.0.2 - 2 sec

### `alienfx-cli setpower`
I used `setone` above because `alienfx-cli lowlevel setpower=0,10,255,0,0,0,0,255` never changed the state of the PB in any version I tried.  But the different versions do take different amounts of time to exit the `setpower` without doing anything - occasionally the same speed as `setone`, often much slower, a couple times faster.  Here's the list:
 * NO  8.0.0 - ran in 3 sec
 * NO  8.0.1 - ran in 3 sec
 * NO  8.0.2 - ran in 3 sec
 * NO  8.1.0.0 - didn't do anything in 20 minutes and I gave up
 * NO  8.1.0.1 - didn't do anything in 2 minutes and I gave up
 * NO  8.1.0.2 - didn't do anything in 2 minutes and I gave up
 * NO  8.1.0.3 - ran in 73 sec
 * NO  8.1.1 - ran in 70 sec
 * NO  8.1.2 - ran in 4 sec
 * NO  8.1.3 - ran in 4 sec
 * NO  8.1.4 - ran in 4 sec
 * NO  8.2.0 - ran in 4 sec
 * NO  8.2.5 - ran in 4 sec
 * NO  8.2.6.0 - ran in 4 sec
 * NO  8.2.6.1 - ran in 4 sec
 * NO  8.2.6.2 - ran in 69 sec
 * NO  8.2.6.3 - ran in 69 sec
 * NO  8.2.6.4 - ran in 69 sec
 * NO  8.2.6.5 - ran in 69 sec
 * NO  8.2.6.6 - ran in 69 sec
 * NO  8.2.6.7 - ran in 69 sec
 * NO  8.2.7 - ran in 69 sec
 * NO  8.3.0 - ran in 69 sec
 * NO  8.4.0 - ran in 69 sec
 * NO  8.4.1 - ran in 69 sec
 * NO  8.5.0 - ran in 60 sec
 * NO  8.5.2.0 - ran in 60 sec
 * NO  8.5.2.1 - ran in 60 sec
 * NO  8.5.2.2 - ran in 60 sec
 * NO  8.6.0.0 - ran in 60 sec
 * NO  8.6.1.0 - ran in 60 sec
 * NO  8.6.1.3 - ran in 60 sec
 * NO  9.0.0.0 - ran in 60 sec
 * NO  9.0.0.1 - ran in 60 sec
 * NO  9.0.0.2 - ran in 60 sec
I've confirmed `alienfx-cli status` shows the correct light marked as a power button.  I have to admit I don't get why there are two colors in the command - I've only ever seen a single color on my power button?

### GUI Crash
I also looked at the history of GUI crashes across these versions.  There are several changes in this behavior as well (but only a single version has behavior changes both here and in the PB `setone` section above: 8.2.6.2)
* the GUI does not crash for versions through 8.1.0.2
* starting in 8.1.0.3 and also in 8.1.1, the GUI crashes 66 seconds after selecting PB color (or after starting, if PB has color assigned before start)
* starting in 8.1.2 through 8.2.6.1, the GUI crashes 3 seconds after starting independent of PB color settings -- it crashes even if PB isn't in any zone
* starting in 8.2.6.2 it goes back to the previous behavior of crashing 66 seconds after selecting PB color (or after starting, if PB has color assigned before start)
* starting in 8.5.0 through 9.0.0.2, it crashes in 56 seconds instead of 66

#### Let me know if there are other tests I can run which would be useful.

T-Troll, 2024-09-21:
Wow! Impressive job, thank you!

Seems like you help me to pinpoint 2 issues:
- I use wrong command for CLI setpower, now should be ok.
- I make an endless recursive loop then trying to set power button for your system. So app crash after stack memory out. Fixed as well.

Here are the [test version](https://github.com/user-attachments/files/17083144/Test.zip), please check.

grghub, 2024-09-21:
> Here are the [test version](https://github.com/user-attachments/files/17083144/Test.zip), please check.

Some clear improvements in this version!
* The gui doesn't crash!  That looks fixed!
* The gui does control all lights including PB when a new color is picked
* `alienfx-cli setpower` now runs in only 2 seconds!

However, there are some things which still aren't behaving as expected:
* neither `alienfx-cli setone=0,10,0,255,0` nor `alienfx-cli lowlevel setpower=0,10,255,0,0,0,0,255` change the PB color (ditto `setall`)
* I've saved a configuration with the PB set to a single Color, but on cold boot the PB color isn't set (it's off).  I tested some previous versions for this, and it worked in 8.0.0 but stopped working in 8.0.1, and hasn't worked since as far as I can tell.  (8.0.0 has other serious issues with color management on cold boot - it has a complicated flashing pattern for all lights - but it was at least able to control all lights)

So overall this test version looks like a good step in the right direction especially for the gui (thank you!), but the cold boot and cli functionality still aren't there yet.

T-Troll, 2024-09-22:
PB REQUIRE 2 colors (AC/Battery). One color will not be saved (it used for fast-pace changes, f.e. effects).
setpower CAN not set color immediately, check after reboot.

Uff... As i told, it's hard to fix this for me - i don't have v2/v3 hardware anymore.

Anyway, try [this](https://github.com/user-attachments/files/17088287/Test2.zip) version (to compare). I use a bit different command chain and MAYBE setpower will now set color immediately.

grghub, 2024-09-22:
> PB REQUIRE 2 colors (AC/Battery). One color will not be saved (it used for fast-pace changes, f.e. effects).
> setpower CAN not set color immediately, check after reboot.

OK, thanks for this vital info, again I had no idea about either of these two important facts.  So in the gui for the PB I must put exactly two colors in the PB-only zone?  And you're saying the second one is not saved?

IMHO this is important info to include in the docs, in the -cli help output, etc. (along with the above fact that v2 hardware supports only 2 colors on cold boot -- so for v2 I have to have exactly 2 colors for PB and 1-2 colors for all others?).  So this invalidates all the `setpower` and gui PB testing I've done, I'll have to redo all of those (rebooting after `setpower`, and listing 2 colors for PB in gui).

Two things worth noting:
* the `alienfx-cli setone=0,10,0,255,0` list of results above (in the "Power Button" section) is I think still valid -- it was indeed able to set the PB color immediately in all the versions where that list says YES.  (so only the second list under the "`alienfx-cli setpower`" heading that says everything is NO is invalidated.)  And that still gives us almost 20 versions where the CLI was able to set the PB color via `setone` - does that give us any clues about which command chains work on the PB?
* In starting to experiment with putting two PB colors in the gui, I see now that up through 8.1.1 the gui actually required 2 colors in the PB zone and only allowed the Power mode (as opposed to Color, Morph, etc); but it loses some of that feature in 8.1.2 (where it still adds/removes the first two colors together but allows more than 2 colors, and the mode it adds them all as has an uninitialized/varying garbage string as the name, and allows mode changes), and loses all of it in 8.2.2.1 (where there's no special PB handling apparent at all for the colors).

> Anyway, try [this](https://github.com/user-attachments/files/17088287/Test2.zip) version (to compare). I use a bit different command chain and MAYBE setpower will now set color immediately.

I'm sorry but `alienfx-cli` in this one still doesn't seem to change any colors - I tried `setone`, `setall`, and finally `setpower` plus cold boot.

In case this info is useful to you (and all this is the same in both Test.zip and Test2.zip): in the gui when I set two PB zone colors (mode "Color"), until a reboot it does set the PB to the first color, but it does the Breath effect on it (morph to black and back).  After reboot no PB color is set, it's just dark.  Also, while it is Breathing, if I run an `alienfx-cli` command the Breathing freezes and it keeps displaying the intensity it was at when `alienfx-cli` ran.

> Uff... As i told, it's hard to fix this for me - i don't have v2/v3 hardware anymore.

Random idea: if there are e.g. 10 ways of doing it, maybe make a testing exe which tries all 10 ways, displaying each one and pausing between them so I can write down the effects, or prompting for a y/n on each ("is the PB red now?", "are all lights white now?") and making a report at the end about what mods work for immediate color change?  Or if a reboot is required after each step, have it prompt me for the # it should try on this run (e.g. 1-10) and do it and then let me reboot between each.

grghub, 2024-09-22:
### `alienfx-cli setpower`
OK, I have a new round of results from running `alienfx-cli lowlevel setpower=0,10,255,0,0,255,255,255` ***followed by a cold boot***.  Still none of them made the specified PB color change on restart, but some versions reset most startup colors to Alienware default aqua, except the right panel lights stayed off (those are marked "NO*"; the rest of the versions made no change to startup colors).
 * NO  8.0.0 - ran in 3 sec
 * NO  8.0.1 - ran in 3 sec
 * NO  8.0.2 - ran in 3 sec
 * NO  8.1.0.0 - didn't do anything in 20 minutes and I gave up
 * NO  8.1.0.1 - didn't do anything in 7 minutes and I gave up
 * NO  8.1.0.2 - didn't do anything in 13 minutes and I gave up
 * NO  8.1.0.3 - ran in 73 sec
 * NO* 8.1.1 - ran in 70 sec - reset most startup lights to Alienware default aqua, except right panel lights are off
 * NO  8.1.2 - ran in 4 sec
 * NO  8.1.3 - ran in 4 sec
 * NO  8.2.0 - ran in 4 sec
 * NO  8.2.6.1 - ran in 4 sec
 * NO* 8.2.6.2 - ran in 69 sec - reset most startup lights to Alienware default aqua, except right panel lights are off
 * NO* 8.2.6.7 - ran in 69 sec - reset most startup lights to Alienware default aqua, except right panel lights are off
 * NO* 8.2.7 - ran in 69 sec - reset most startup lights to Alienware default aqua, except right panel lights are off
 * NO* 8.4.1 - ran in 69 sec - reset most startup lights to Alienware default aqua, except right panel lights are off
 * NO  8.5.0 - ran in 60 sec
 * NO  8.5.2.2 - ran in 60 sec
 * NO  8.6.0.0 - ran in 60 sec
 * NO  9.0.0.2 - ran in 60 sec
 * NO  9.0.0.2+-test - ran in 2 sec
 * NO  9.0.0.2+-test2 - ran in 3 sec

### PB color on cold boot
So far, of all the versions and methods I've tried, the only way I can get a specific color to show up on the PB at cold boot is to save from the gui in 8.0.0 (and then it does its complicated flashing pattern with multiple colors and lights...)

T-Troll, 2024-09-23:
Being honest, i'm a bit tired of this fight, then i can't check result.
So... I found PCAP intercept between mine 13R2 hardware and AWCC.
I'm seriously rebuilt SDK to follow it byte-to-byte.

So if [this](https://github.com/user-attachments/files/17094832/test3.zip) version will not work correctly - i give up UNTIL you can provide me PCAP between YOUR hardware and AWCC.