T-Troll/alienfx-tools #448: AlienFX Monitor isn't updated?

neoacevedo, 2024-09-01:
I recently downloaded the 9.0.0.0 MSI version of alienfx-tools. The AlienFX Monitor and AlienFan Control are updated to that version number, but when I openopening AlienFX Monitor it launches the new version update notification and when I check the about menu, this is in the previous version.

[image]
[image]
[image]

T-Troll, 2024-09-02:
I check - MSI have correct version for all tools.
Your situation can be met if some tools run during update and install can't stop it.
So try to stop all tools and run msi again.

neoacevedo, 2024-09-08:
@T-Troll it didn't work. 

I didn't have any AlienFX tool running and I disabled the Windows Defender to check if it was doing something. I had to uninstall the software and install it again to get the latest version for AlienFX Monitor.

closed 2024-09-13 by T-Troll