T-Troll/alienfx-tools #451: GMode/High Performance Mode unavailable on x15 R2 BIOS 1.22

Pitagoras17, 2024-09-10:
**Describe the bug**
On v. 9.0.0.1 the option for GMode is not available. It's unresponsive to hotkeys and greyed-out on the gui. This seems to also affect the power modes, since they are only listed from "Manual" to "Level 163-171". If selected, Level 171 (Highest available), makes the GMode (F1 key) flash and freezes/crashes ensue.

**System (please complete the following information):**
 - Model: Alienware x15 R2
 - Model details: Intel Core i9-12900H, 32GB, RTX 3080 Ti, BIOS 1.22
 - OS: Windows 11 23H2

**Additional context**
This system is also unresponsive to the performance fan option in BIOS, or any of the AWCC Thermal Profiles. The only way to change fan speed is thorugh manual control.

WilfulTravesty, 2024-10-03:
I have the same laptop with the same CPU/GPU/Bios. While I'm on AlienFX-Tools v9.0.1 now, I've been running the last few versions of AlienFx-Tools, and I don't think I've ever seen the F1 key light up when in mode 171/G-Mode, and I have always been able to select it from the GUI. It did light up when running Alienware from Dell, so I guessed it was a function of that software.

The first time I enter mode 171 after waking up I often see a "pause", but it only lasts for 2-3 seconds, and it has always done so, even when running the Alienware software from Dell. 

I write this just for additional information.

BTW, for reference I added power measurements for modes 160-171 to issue #400

T-Troll, 2024-10-04:
Duplicate of #400

closed 2024-10-04 by T-Troll