T-Troll/alienfx-tools #454: Dell G16 7630 XMP Issue - bricked mobo

Blackwolf14, 2024-09-19:
I've tried Xmp 2 but my mobo bricked. Fan spinning, keyboard lighting but no picture at all. I made a huge mistake.

Blackwolf14, 2024-09-19:
I've fixed it.

-Reseat ram modules
-drain flea power.

Those worked for me.

Be aware, don't change xmp settings on dell g16 7630. Xmp should be set to 0 on dell g16.

T-Troll, 2024-09-21:
Seems like i detect XMP there is not available.
Your BIOS has functions for this, but seems like it hangs the system instead.

Can you please provide me the result of `alienfan-cli dump` command and, also, ACPI dump (if possible)?

Blackwolf14, 2024-09-21:
[image]
I didn't find acpi dump. I tried this command `./alienfan-acpi dump` but no luck.