T-Troll/alienfx-tools #455: Alienware m16 R1 - keyboard ABNT2 - Brasilian layout

amarildolsouza, 2024-09-21:
**Describe the bug**

Device:  VID_0D62/PID_D2B0, 90 lights, APIv5

Key "\" + "|" is not mapped.
Key "]" + "}" is not mapped.
Key "Ç" is equal to ";".
Key "/" is  equal to "Right Control".

Photos of keyboard:
[image]
[image]

Screenshot of AlienFX Control
[image]

Thanks for any help.

Have a great day.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
 - Model: [e.g. AlienWare m15R3]
 - Model details: [e.g. per-key rgb keyboard]
 - OS: [e.g. Windows 11 stable]

**Additional context**
Add any other context about the problem here.

T-Troll, 2024-09-22:
Yes, i have English (well... not exactly) keyboard, layout is different.
Can you please edit mappings and grid, then share me your version using "Save lights" button?
I'll add it as m16R1 (Brazil).

amarildolsouza, 2024-09-22:
> Yes, i have English (well... not exactly) keyboard, layout is different. Can you please edit mappings and grid, then share me your version using "Save lights" button? I'll add it as m16R1 (Brazil).

Yes, off course.
Amazing.

Thanks!

---

Lights and grids:
[Alienware m16R1 (Brazil).csv](https://github.com/user-attachments/files/17089821/Alienware.m16R1.Brazil.csv)

amarildolsouza, 2024-10-02:
> Yes, i have English (well... not exactly) keyboard, layout is different. Can you please edit mappings and grid, then share me your version using "Save lights" button? I'll add it as m16R1 (Brazil).

Hello!

If you have finished the file for the ABNT2 Brazil keyboard layout, can you please make it available?

Thanks!

T-Troll, 2024-10-02:
Ok. it will be included into next release.