T-Troll/alienfx-tools #456: Way to disable TCC - I use ThrottleStop for this

amarildolsouza, 2024-09-22:
**Is your feature request related to a problem? Please describe.**
The TCC feature is overriding the same option in ThrottleStop.

**Describe the solution you'd like**
Option to disable TCC control, I use ThrottleStop for this.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**System (please complete the following information):**
 - Model: [e.g. AlienWare m15R3]
 - Model details: [e.g. per-key rgb keyboard]
 - OS: [e.g. Windows 11 stable]

**Additional context**
Add any other context or screenshots about the feature request here.

T-Troll, 2024-09-22:
Well... First of all, TCC can not be disabled - it's an internal feature of the CPU.
It can be reset to default value for CPU (Usually 100C for Intel, 105C for AMD).
Second point - me and TS set it using different ways - they access CPU registers directly, i'm using Dell ACPI BIOS functions for this. 
Third point.... At my gear (m16 as well), TS provide significant performance drop, so i stop using it. JFYI.

So, keep this points in mind, what do you want to have as a result?

amarildolsouza, 2024-09-22:
> Well... First of all, TCC can not be disabled - it's an internal feature of the CPU. It can be reset to default value for CPU (Usually 100C for Intel, 105C for AMD). Second point - me and TS set it [...]

By "Off" I mean "Fan Control" stop control TCC and let, for example, ThrottleStop control it.

But I will follow your advise and will disable on ThrottleStop and will 
test on Fan Control.

closed 2024-09-25 by amarildolsouza