T-Troll/alienfx-tools #460: Alienware m16 R1, Intel Graphics + RTX 4070, GPU nVidia temperature is not accurately.

amarildolsouza, 2024-09-25:
**Describe the bug**
"AlienFX Fan Control" and "AlienFX Monitor" are not showing/getting temperature accurately.

The programs may be getting the temperature from the Intel UHD Graphics card.

**GPU-Z for Intel Graphics**:
[image]

**GPU-Z for nVidia RTX 4070**:
[image]

**Windows 11 Pro Task Manager**:
[image]

**Geeks3D FurMark**:
[image]

**AlienFX Fan Control**:
[image]

**AlienFX Monitor**:
[image]

Thanks for any tip.

Have a nice day.

---

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
 - Model: [e.g. AlienWare m15R3]
 - Model details: [e.g. per-key rgb keyboard]
 - OS: [e.g. Windows 11 stable]

**Additional context**
Add any other context about the problem here.

T-Troll, 2024-09-25:
No, it's correct.
Dell use their own sensors for CPU/GPU, not the internal ones. So temperature can vary.
Some internal (and additional sensors from MB) exposed as "ESIF sensors" (but Intel do not provide exact name until some trick done).

amarildolsouza, 2024-09-25:
> No, it's correct. Dell use their own sensors for CPU/GPU, not the internal ones. So temperature can vary. Some internal (and additional sensors from MB) exposed as "ESIF sensors" (but Intel do not provide exact name until some trick done).

AlienFX Fan Control relies on this sensor to trigger the GPU Fan more accurately.

Thanks for the quick response.

T-Troll, 2024-09-25:
Well... In my tools, you can use ANY sensor (or sensor combination) to control ANY fan. So just configure out what is better for you needs.
f.e me, for m15R1, use SSD sensor for GPU fan (SSD is close to GPU, so it needs to be cooled if hot using GPU fan).

amarildolsouza, 2024-10-08:
There is other users with same situation.