T-Troll/alienfx-tools #461: AW3423DWF not detected

panki27, 2024-10-01:
Hi,

I just setup alienfx-tools and tried it out, but my AW3423DWF is not detected.

```
$ alienfx-cli.exe probe
alienfx-cli v9.0.1
Dell API ready, 1 devices found.
===== Device VID_187c, PID_100e =====
+++++ Detected as: Realtek HID Device, APIv9 +++++

$ alienfx-cli.exe status
alienfx-cli v9.0.1
Dell API ready, 1 devices found.
Device #0 - Realtek HID Device, VID#6268, PID#4110, APIv9, 0 lights
Done.
```

Anything else I can do to debug this? Thanks in advance :)

T-Troll, 2024-10-02:
Unfortunately, APIv9 (dwF monitors) is still under development. It detected into my tools (as you can see), but can't be controlled yet.
Please wait for updates.

closed 2024-10-02 by T-Troll

panki27, 2024-10-02:
Thank you for taking the time to answer, if you need help in researching this let me know.

T-Troll, 2024-10-02:
They add some kind of encryption into commands.
Now we have international group hacking it, but this took time.
Ask @Fefedu973 for details.

TheMasterDingo, 2024-10-18:
Is there any chance for this monitor to be supported in the near future? Thanks

T-Troll, 2024-10-22:
Yes, we are working on it, but no forecast about dates, sorry.

TheMasterDingo, 2024-10-22:
Cool, thanks

Fefedu973, 2024-10-22:
The reverse engineering work has been done, we now need to implement it into the software in a good way

t56k, 2025-03-06:
Hi all, any update here? I get overheating warnings on NixOS, I suspect the fan isn't spinning up

Fefedu973, 2025-03-06:
It is supported in openrgb and Signal rgb btw
(You just need my plugin)