T-Troll/alienfx-tools #462: Key colors missing after hibernation

WilfulTravesty, 2024-10-03:
**Describe the bug**
Every time I wake my laptop from hibernation, four keys have no color. Selecting "Refresh" in AlienFX-Tools restores the colors.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure lighting on all keys (either dim or full brightness).
2. Minimize AlienFX-Tools to the system tray.
3. Hibernate the computer.
4. Reflect on the joy of having a gaming computer.
5. Wake the computer.
6. Observe the keyboard performing a rainbow color walk while booting, with all keys lighting up.
7. Notice that during the Windows 11 loading screen, all keys display the expected color.
8. As the Windows 11 login screen appears, observe that keys F1, F12, Speaker Mute, and Microphone Mute are unlit.
9. Unlock the computer, then "Refresh" AlienFX-Tools from the system tray and notice that the colors are now correct.

**Expected behavior**
The affected keys should retain the programmed colors like the rest of the keyboard after logging in/unlocking, without needing a manual refresh.

**System:**
 - Model: Alienware x17 R2
 - Model details: per-key RGB keyboard, Dell Technologies Keyboard, VID_0D62/PID_BABC, 89 lights, APIv5
 - OS: Windows 11 (fully patched)
 - Antivirus: Bitdefender Total Security

**Additional context**
Note that 3 of the problematic keys are system-controlled:

- F1 is the G-Mode key with Alienware software, which lights up when G-Mode is enabled, but when using AlienFX-Tools it does not change when the mode is set to 171.
- Speaker Mute turns white when the system mutes the sound, and reverts to the set color when unmuted.
- Microphone Mute also turns white when muted, and reverts to the set color when unmuted.

- F12 is a standard key. It can disable the touchpad, but its color does not change when this happens.

T-Troll, 2024-10-04:
Is it always the same buttons or random?

WilfulTravesty, 2024-10-04:
It’s always the same 4 buttons that go dark: F1, F12, Speaker Mute, and Microphone Mute.

Another note, in case it's relevant: the only other key on the keyboard (besides F1, Speaker Mute, and Microphone Mute) that automatically changes color when pressed is the key to the right of the spacebar, which disables the Windows key (located to the left of the spacebar). This key also turns white when the Windows key is disabled, but it always displays the correct color after waking up.

WilfulTravesty, 2024-10-05:
Speaker Mute is light 16.
Microphone mute is light 19.
F1 is light 1
F12 is light 12

Unused values which don't light up any keys when selected on the Devices and Grids tab: 
33, 36-39, 41, 54, 56-59, 73, 75-79, 82, 93, 95-99, 106, 107, 108, 110, 113, 115, 116-132

I tried enabling speaker and microphone mute by pressing the keys so they lit up white, then stepping through all the keys on the Device and Grids tab to see if any would light up (green), but they did not and just remained white. When I pressed them again to un-mute, they lit up at their regular light ids (16 and 19, above).

Note that the colors load fine after a reboot and log in. It's just after a hibernate, wake, and log in that they are unlit.

WilfulTravesty, 2024-10-30:
I checked the Device effects setting you mentioned in your last response, but it was already off.

It looks like the latest update fixed the problem because it hasn’t happened again since installing it. I'll close this issue.

Thank you!

closed 2024-10-30 by WilfulTravesty