T-Troll/alienfx-tools #463: Wrong temps readings for the GPU

Bakr228, 2024-10-06:
Wrong temps reading for GPU

the linked picture shows the difference between AlienFX readings and HWINFO and RIVA Tuner readings.
 is it a bug or something else?

[image]

T-Troll, 2024-10-06:
Something else.
First 2 sensors is Dell's proprietary (AWCC using it for control as well).
And yes, it's not a first time it provides wrong settings (by design or by BIOS bug).

What to do?
Use other (ESIF) sensors to control the fans. If GPU not found across ESIF (MB design can not provide this info), try to run Libre Hardware Monitor before my tools - it's sensors supported as a data source.

T-Troll, 2024-10-06:
Following our discussion at GitHub - try this - [alienfx-gui.zip](https://github.com/user-attachments/files/17268555/alienfx-gui.zip)
Let me know if:
1. It solves sensor readings delay
2. It not provide stutter under heavy load

Bakr228, 2024-10-06:
[image]
the problem is still there no stutter and I waited around 10 min before the screen shot,
 about other ESIF I can't find an ESIF sensor close to the actual readings of the GPU.

Bakr228, 2024-10-07:
[image]
Why did you send me a trojan ?

amarildolsouza, 2024-10-08:
I have the same issue.

The temperature reading is from Intel Graphics.

https://github.com/T-Troll/alienfx-tools/issues/460

T-Troll, 2024-10-10:
@Bakr228 It's well-known false positive. M$ have no feedback to resolve. I even have wiki page about it.

T-Troll, 2024-11-10:
Heh! TS have External GPU temp at ESIF 2.
And no, AWCC temp is not iGPU one - it uses MB-embedded sensor. But can be buggy sometimes.