T-Troll/alienfx-tools #464: Fan control not working on X14R2?

dynamiszmx, 2024-10-11:
The fan curve in manual mode behaves oddly on my Alienware X14R2. I've set a flat line with a boost of 10 as long as the temperature stays below 80 degrees, but the fan still speeds up as the temperature rises.

The boost remains constant at 10, but the actual fan speed—the percentage displayed next to the boost number—still increases. Am I misunderstanding how this function works?

[image]

System 
 - Model: Alienware X14 R2
 - OS:  Windows 11

T-Troll, 2024-10-14:
Yes, this is correct.
We only have indirect fan control for our gear, so boost can only increase RPM above BIOS settings (but still keep it not so loud).

Use power modes for quiet operation (but it will cost lower TDP limits).

dynamiszmx, 2024-10-14:
thank you!

> Yes, this is correct. We only have indirect fan control for our gear, so boost can only increase RPM above BIOS settings (but still keep it not so loud).
> 
> Use power modes for quiet operation (but it will cost lower TDP limits).

closed 2024-10-14 by dynamiszmx