T-Troll/alienfx-tools #478: AlienFX GUI Opens then Closes

JazzyJazzjazzy, 2024-12-18:
**Describe the bug**
AlienFX GUI Opens then Closes. Unable to open the program window.

 - Model: [e.g. AlienWare m18R1]
 - Model details: [e.g. per-key rgb keyboard]
 - OS: [Windows 11 stable]

T-Troll, 2024-12-19:
Ohh... I know some setting combination provide this, but can't pinpoint it yet.
Can you, please, backup your settings using `alienfx-config` and share here?

JazzyJazzjazzy, 2024-12-20:
Apologies, but I will not be able to access my computer for a few weeks. I will post back then. Thank you.

detritivore, 2024-12-24:
I've been having the same issue (or at least the same symptom, ie. crash on start of the app) since version 9.0.3. Just tried with the new 9.1.0, same problem.
9.0.2 works fine.

I'm on an Alienware m15 R3 (2020)

[alienfx-config-backups.zip](https://github.com/user-attachments/files/18237975/alienfx-config-backups.zip)

T-Troll, 2024-12-24:
Thanks, i'll take a look.

T-Troll, 2024-12-30:
@detritivore, I can't reproduce it for now, works well at mine configuration.
Seems like something wrong with color settings, and i found one crash about it into my code already, but don't sure it's about this.
So i need to wait @JazzyJazzjazzy - i have the same hardware as him, will be much easier to pinpoint.

JazzyJazzjazzy, 2024-12-30:
Hi, saw your message. I can have the file sent in my absence. Can you send me the steps to complete to get the file. Thank you.

T-Troll, 2024-12-30:
@JazzyJazzjazzy,

1. Open CMD into tools folder (need Admin only if installed in Program Files).
2. Issue command `alienfx-config backup) - it will create 3-4 .reg files into the same folder.
3. Pack it into .zip and attach here (or send me in Discord as .reg).

JazzyJazzjazzy, 2024-12-30:
Here 
[Reg 1.zip](https://github.com/user-attachments/files/18276497/Reg.1.zip)
are .reg files.. Sorry for delay.

T-Troll, 2024-12-31:
Got it!

You assign the zone with no-grid lights into Ambient grid effect. My app can't handle this situation (it have zero-size ambient grid in this case). 

Of course, crash is not an option, so i will just turn off effect into this case for the next release.

PS: In case you need out-of-grid lights to participate into Ambient effect, use Ambient tab instead. It can set all zone to pre-defined Ambient grid points. You can mix both Ambient (from tab and from grid effect), it can work together.

JazzyJazzjazzy, 2025-01-01:
The reason that I assigned the zone with no-grid lights into Ambient grid effect was that no lights were lighting up at all. I was just trying to get the lights to work.

T-Troll, 2025-01-02:
Oh. I see you have solid color activated, but Ambient can override it.

detritivore, 2025-01-26:
hey, just wanted to let you know, I just installed version 9.1.1 (I skipped the previous ones since 9.0.2), and it now no longer crashes!
Cheers :-)

T-Troll, 2025-01-29:
Thank you for the information.

closed 2025-01-29 by T-Troll