T-Troll/alienfx-tools #482: AlienFx Fan control triggers WMI activity error when plug in

dynamiszmx, 2025-01-04:
**Describe the bug**

Observe the Task Manager and notice that Windows Management Instrumentation (WMI) and the WMI Provider Host process are triggered and continuously report activity error(reported by alienfx tools).
Seems only happen when plug in, issue does not exist when using battery. Power does doesn't matter.
AWCC and all other DELL apps uninstalled. Only keep Alienfx tools.

**To Reproduce**
1. Open Settings in the application.
2. Enable the option "Enable Fan Control".
3. Observe the Task Manager and notice that Windows Management Instrumentation (WMI) and the WMI Provider Host process are triggered and continuously report activity.
4. Disable "Enable Fan Control", which resolves the issue and stops the WMI activity.
5. Laptop plug in.

**Expected behavior**

Report detail here on Event Viewer.
Log Name:      Microsoft-Windows-WMI-Activity/Operational
Source:        Microsoft-Windows-WMI-Activity
Date:          2025/1/4 16:39:26
Event ID:      5858
Task Category: None
Level:         Error
Keywords:      
User:          SYSTEM
Computer:      Dynam14r2
Description:
Id = {00000000-0000-0000-0000-000000000000}; ClientMachine = DYNAM14R2; User = Dynam14r2\dynam; ClientProcessId = 12076; Component = Unknown; Operation = Start IWbemServices::ExecMethod - ROOT\WMI : \\DYNAM14R2\ROOT\WMI:AWCCWmiMethodFunction.InstanceName="ACPI\\PNP0C14\\AWCC_0"::Thermal_Information; ResultCode = 0x80041001; PossibleCause = Unknown
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Microsoft-Windows-WMI-Activity" Guid="{1418ef04-b0b4-4623-bf7e-d74ab47bbdaa}" />
    <EventID>5858</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>0</Task>
    <Opcode>0</Opcode>
    <Keywords>0x4000000000000000</Keywords>
    <TimeCreated SystemTime="2025-01-04T21:39:26.0671958Z" />
    <EventRecordID>8916</EventRecordID>
    <Correlation ActivityID="{3b6fbe94-5eee-0005-75bf-6f3bee5edb01}" />
    <Execution ProcessID="4544" ThreadID="5876" />
    <Channel>Microsoft-Windows-WMI-Activity/Operational</Channel>
    <Computer>Dynam14r2</Computer>
    <Security UserID="S-1-5-18" />
  </System>
  <UserData>
    <Operation_ClientFailure xmlns="http://manifests.microsoft.com/win/2006/windows/WMI">
      <Id>{00000000-0000-0000-0000-000000000000}</Id>
      <ClientMachine>DYNAM14R2</ClientMachine>
      <User>Dynam14r2\dynam</User>
      <ClientProcessId>12076</ClientProcessId>
      <Component>Unknown</Component>
      <Operation>Start IWbemServices::ExecMethod - ROOT\WMI : \\DYNAM14R2\ROOT\WMI:AWCCWmiMethodFunction.InstanceName="ACPI\\PNP0C14\\AWCC_0"::Thermal_Information</Operation>
      <ResultCode>0x80041001</ResultCode>
      <PossibleCause>Unknown</PossibleCause>
    </Operation_ClientFailure>
  </UserData>
</Event>

**Screenshots**
Attached.
task manager with fan control on and off.  Error on Even viewer

[image]
[image]
[image]

**System (please complete the following information):**
 - Model: [e.g. AlienWare X14R2]
 - OS: [e.g. Windows 11 23H2]

**Additional context**

T-Troll, 2025-01-05:
Well... I have not so good news for you.

Yes, all fan and power control utilize WMI calls to proprietary Dell methods, so WMI activity in itself is normal.
But you have an error to ThermalInformation retrieval - it's internal system thermal/fan/power sensors' data. So, seems like, it was not mapped correctly to ACPI BIOS calls.

This error can appear once at app's start (during hardware detection process), but in case it repeats during operation - your WMI data on Windows is broken and need to be repaired (by Windows clean install).

Did "Fans and Power" operating correctly?

dynamiszmx, 2025-01-05:
Yes, the power mode  is fully functional. I can switch between mode 160 to 164 as well as Manual and G-Mode, everything is working as expected.

[image]

I can see the GPU and CPU sensors, along with some ESIF sensors. Thanks for your explanation regarding WMI—it makes sense now. The WMI is triggered every time the app attempts to poll the fan control, regardless of the power mode(now its 1000ms).

As a temporary solution, I’ve increased the polling period to 10,000,000ms, which helps reduce the frequency of WMI activity. Additionally, I will not use a custom fan curve.

I don’t have AWCC or Dell Core Service installed. Are they necessary for proper functionality?
And the requirements says:
Alienware light device/Alienware ACPI BIOS (for fan control) present into the system and have USB HID driver active (alienfx-cli can work even with missing devices, Dell LightFX needs to be present in the system).

> Well... I have not so good news for you.
>
> Yes, all fan and power control utilize WMI calls to proprietary Dell methods, so WMI activity in itself is normal. But you have an error to [...]

T-Troll, 2025-01-05:
> As a temporary solution, I’ve increased the polling period to 10,000,000ms, which helps reduce the frequency of WMI activity.

It can help, but keep in mind - temperature sensors update will be at the same rate.

> I don’t have AWCC or Dell Core Service installed. Are they necessary for proper functionality?

No. As i write into readme, only BIOS with these functions needed - it's exposed to WMI by Windows PnP (see - `InstanceName="ACPI\PNP0C14\AWCC_0"::Thermal_Information` - from you logs).
For lights, no specific driver (except MS default "Vendor-defined HID device") required as well.