T-Troll/alienfx-tools #484: Alienware Fan Control not showing up on notification bar after restart

Danfun64, 2025-01-10:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Install AlienFX-Tools
2. Run Alienware Fan Control
3. Have "Start with Windows" and "Start minimized" enabled
4. Restart PC
5. Notice icon missing from notification icons

**Expected behavior**
I use Alienfan Control as a means to turn G-Mode on and off without the bloated AWCC installed. Usually, I could tell from the icon whether g mode is enabled or not. However, recently it appears the program runs but doesn't show in the notification bar for some reason unless I close it with task manager and reopen it again. The problem started today and happens consistently whenever I restart my PC.

**Screenshots**
[image]

**System (please complete the following information):**
 - Model: Dell G15 5520
 - Model details: [e.g. per-key rgb keyboard]
 - OS: Windows 11 24H2

**Additional context**
Add any other context about the problem here.

Danfun64, 2025-01-11:
If I use it with Start Minimized turned off, it shows up at startup as expected but the notification icon is still hidden.

ivanmestre, 2025-01-14:
@Danfun64 I had the same issue, and found a workaround. I opened the Windows Task Scheduler and added a 30 seconds delay to the Trigger of the task created by AlienFX. So far, it has worked perfectly since

[image]

T-Troll, 2025-01-18:
Seems like W11 doesn't have event for taskbar ready (W10 have).
It's quite strange for me, because i didn't meet this issue at my systems.

Anyway, i'll do some fix for the next release.

Danfun64, 2025-01-25:
Seems to be working for me now with the new release. Thanks!

closed 2025-01-25 by Danfun64