T-Troll/alienfx-tools #485: Dell g15 5515 ryzen edition (system ID: 3200) still gets fan stuck.

Elhinko, 2025-01-14:
Fans still get stuck at a single RPM and don't change even if the CPU hits max temp.
This issue has been happening since I bought the laptop even with AWCC, throughout all releases of windows till 24H2 so far.

There was a patch for this in release 9.0.0.2 but it sadly doesnt work
> All fan-control tools:
> Override added for SystemIDs 4800 and 3200 to prevent fan stuck. 
[release 9.0.0.2](https://github.com/T-Troll/alienfx-tools/releases/tag/9.0.0.2)

fans get stuck after idling for 15-30 minutes usually, only way to get them unstuck is G mode on then off.

Attached is a screenshot of CPU fan being stuck at 400 RPM even though CPU is 92c° and curve is max boost.
[image]

Elhinko, 2025-01-15:
So I was just looking through other issue tickets and it seems like this is a common dell bug with some models.
[Ticket 344](https://github.com/T-Troll/alienfx-tools/issues/344) [Ticket 417](https://github.com/T-Troll/alienfx-tools/issues/417)

I tried multiple bios versions but none of them seemed to fix the issue.
I also noticed that fans don't hang if I am doing intensive work on the laptop and they are spinning fast, only when the system is idle/I am just watching youtube.

It's insane how dell can't get their firmware/software working. From AWCC being bloated crap that fails to install. to bios that cant even spin the fans properly. with missing features like advanced optimus (Even though my 5515 has a 3060 and a mux switch).
would have never bought this laptop if I knew dell was this bad.

gfnord, 2025-01-18:
After the last BIOS update to version 1.9.0 on my Alienware M16 R2 the fan control stopped working. :(

T-Troll, 2025-01-18:
Being honest, i'm a bit tired to fight buggy 5515/5525 BIOS (well... seems like it not BIOS, but SPI firmware - boost set correctly, but not activate fans).

For now, i can only recommend do not use manual mode for IDLE, but any other, and switch to manual mode using app detection (profiles into Alienfx-GUI).

@gfnord How does it look like?

Elhinko, 2025-01-18:
@T-Troll So I tried the only other power mode I have (level 151) and 1 minute after temps drop below 45-50c both fans stop spinning and the don't start spinning again unless g-mode on/off.

[image]

I also wanted to ask you if you know about a way to reach out to dell to try to get them to fix these issues.

T-Troll, 2025-01-19:
AFAIR, you should first switch to mode 0xAB (171), then to any other. This should fix the situation.

gfnord, 2025-01-19:
Sorry about my ignorance, but how to switch to mode 0xAB? Has the M16 R2 this mode?

T-Troll, 2025-01-19:
@gfnord it should be marked as "Level 171". It can be absent for some models, but we are talking about a specific one. Don't sure about R2, but R1 have it (and works well without this workaround).