T-Troll/alienfx-tools #487: Alienware m16 AMD - Set Fans to RPM or Percent

Wromper, 2025-01-18:
Thanks for putting so much time and effort into the software.  I have used it for lighting, but for the life of me, cannot figure out how to use it to control the fans.  By default, all 4 fans like to spin up at low temps, causing a lot of unnecessary noise.  I would like to simply set the fans to 50%, or even change their top speed from 100%-130% down to 70%.

I have tried the GUI and the CLI and cannot seem to find the right combination of commands.  The readme file doesnt show a lot of examples.  Can you please tell me how to forceably set the fan speed, or set a max %?  

thanks!

T-Troll, 2025-01-19:
There are two things about fan control at our hardware:

- Fan control is INDIRECT. You can't control real speed, only provide a hint for pre-defined profile.
- Fan can only be controlled in Manual mode.

So... You can't do this way. And it's for good, in fact, BIOS keep eye to prevent overheat (fans to 0 at high temp).
But you can use different power modes for IDLE/mid/high load. I have profiles mechanic into FX GUI for automatic mode switch.

Wromper, 2025-01-19:
Thanks for the reply.  I have been in IT for a long time and understand
that messing with the fans can be dangerous.  I am not trying to turn them
off, just want to control them better.  The fan noise is very distracting
and IMHO, unnecessary to spin them at 100% when the laptop is at 68
degrees. One of the fans spin all the way up to 5000 rpm, which emits a
whining noise.

You mention manual mode, but how do I do that in the software?  I can do it
in Dell Fan Command by Aaron(?), because it has % option to manually run
the fans.  But that software only sees 2 of the 4 fans.  There is very
little cooling difference between 50% and 100%, even under heavy load and
stress tests.

Can I set the fans to *anything *manually? If so, I don't know how to use
the manual option.
CPU3 fan seems to be the 5000 rpm fan.  It might be called the Chipset Fan
in the Dell tools.  It is physically smaller than the other fans, so I
guess Dell decided it needed to spin faster.

thanks, R