T-Troll/alienfx-tools #490: Dell G16 laptop: none of the tools load with Windows startup

boss281, 2025-01-30:
**Describe the bug**
Despite the option checked to load the tool(s) during startup, none of the tools will load automatically.

**To Reproduce**
In Settings, select "Start With Windows"

**Expected behavior**
The tool to load and show up in the taskbar tray

**System (please complete the following information):**
Dell G16 7630
Windows 11 stable

T-Troll, 2025-01-31:
This can be because i change installation folder at latest release.
You can open Task Scheduler and check AlienFX-GUI/AlienFan-GUI entry (Actions tab) to check what path it used.
In case the path is incorrect, just turn off start-up loading in tools settings, then turn it back on - this should fix the situation.

boss281, 2025-01-31:
T-Troll, there are no entries for AlienFX in task scheduler.

T-Troll, 2025-01-31:
Interesting... Did switching it off-on in app create the entry?
PS: You may need to run app from Admin to do so at some systems.

boss281, 2025-01-31:
No, unfortunately switching it off/on did nothing for the alienfan or alienfx. Neither did running it as admin. No entries in Task Scheduler.

closed 2025-01-31 by boss281