T-Troll/alienfx-tools #491: Lights Will Not Come On

JazzyJazzjazzy, 2025-02-01:
Lights Will Not Come On no matter how I configure AlienFx. I have no long explanation, simply cannot get any lights to illuminate. Is it me or the software?

T-Troll, 2025-02-02:
Do you have m16R1, right?
In this case, something wrong with your setup, i use the same hardware without issue.

JazzyJazzjazzy, 2025-02-02:
I have m19R1. I thought it was probably just me but I cannot find my error.

[Reg.zip](https://github.com/user-attachments/files/18631681/Reg.zip)

JazzyJazzjazzy, 2025-02-06:
Sir,

I have tried everything to get these lights to work again. Maybe my BIOS update caused the issue. I have NO IDEA. The only lights that respond are those of the TRON and VENTS on the rear of the display. Keyboard remains unresponsive to any input from AlienFx. Thank you.