T-Troll/alienfx-tools #498: x16 R2 - alienfx-gui does not open

jeffreyyu0406, 2025-02-22:
**Describe the bug**
Upon installation, alienfx-gui does not open. All other tools open as normal.

**To Reproduce**
Steps to reproduce the behavior:
1. Open alienfx-gui - process opened on Task Manager, but immediately closes.

**Expected behavior**
AlienFX GUI should open as expected.

**System (please complete the following information):**
 - Model: Alienware x16 R2
 - Model details: NVIDIA RTX 4070 Laptop, Intel Core Ultra 9 185H, Per-Key RGB Keyboard
 - OS: Windows 11 24H2 Version 10.0.26100 Build 26100

T-Troll, 2025-02-23:
Interesting...

Can you please:
1. Open CMD
2. Run `alienfx-cli probe` and share output (answer "n" to it's question)?
3. Run `alienfx-config backup` and share .reg files it create?

jeffreyyu0406, 2025-02-24:
I tried running `alienfx-cli probe` on a admin-level cmd, but it was not recognized as an internal or external command.

T-Troll, 2025-02-25:
It doesn't require admin, but the command should be issued from the folder you install/unpack my tools.

jeffreyyu0406, 2025-02-25:
Gotcha, please see attached.

[image]

[afx reg files.zip](https://github.com/user-attachments/files/18974256/afx.reg.files.zip)

T-Troll, 2025-02-26:
Got it! It's a bug into Global Effects.
I'll fix it in an upcoming release.

Turoxit, 2025-06-23:
Interesting. On my side the launch is ok, the software run. 
But he detect any key... i' have to map all the computer? 

[image]

T-Troll, 2025-06-24:
@Turoxit Yes. 
Unfortunately, there is no way to extract all lights from the chip, and i'm not the Dell so i don't have specs for all models.
Some people create mappings and share it for me - you can use "Detect devices" button to see if some scheme fit your hardware.
DO NOT use "similar" models, yours exactly only!
If it not presents... Read [beginner's guide](https://github.com/T-Troll/alienfx-tools/wiki/How-to-start-%28Beginner%27s-guide%29) about how to set up it manually. It's a one-time process.

afatdeafkid, 2025-07-23:
Hello I am having the same issue, cmd output is as follows

alienfx-cli v9.2.0.1
Dell API not found, 0 low-level devices found.
Light devices not found, exiting!

I have my alienware keyboard on and I have alenware command center running as well

T-Troll, 2025-07-25:
What model of keyboard do you have?

afatdeafkid, 2025-07-25:
Alienware pro wireless. I'll post the link below

https://www.dell.com/en-us/shop/alienware-pro-wireless-gaming-keyboard/apd/580-bbmv/pc-accessories

afatdeafkid, 2025-07-25:
Ah ok, sounds good. I'll run through that process tomorrow. Thank you