T-Troll/alienfx-tools #501: Alienware m16R1 (Brazil)

amarildolsouza, 2025-03-02:
**Describe the bug**
After version 9.1.3.0 this two keys are off: "\, |" and "], }".

**Screenshots**
[image]

**System (please complete the following information):**
 - Model: [AlienWare m16R1 (Brazil)]
 - Model details: [per-key rgb keyboard with Brazilian layout]
 - OS: [Windows 11 Pro 23H2]

**Additional context**
I did a clean install and nothing solve.
I started config from scratch and no success.

T-Troll, 2025-03-02:
First, make sure you have correct keyboard type in BIOS(!). Yes, some mappings depend on this.
Second, make sure you are use "Brazil" version at detection, not US one.

In case this is not helped, you need to remap it manually, using manual lights setup to find correct IDs for this lights.

amarildolsouza, 2025-03-02:
> First, make sure you have correct keyboard type in BIOS(!). Yes, some mappings depend on this. Second, make sure you are use "Brazil" version at detection, not US one.
> 
> In case this is not helped, you need to remap it manually, using manual lights setup to find correct IDs for this lights.

Yes, correct keyboard type in BIOS.

[image]

Correct version:

[image]

amarildolsouza, 2025-03-02:
How can I solve this and probably help other owners of the same laptop model?

How to remap it manually?
Using manual lights setup to find correct IDs for this lights?

dc-IV-AW, 2025-03-02:
@amarildolsouza , have you verified you have a Light Zone with those keys assigned? Here is my US m18 R1, so I don't have those keys, but I did assign all "Punctuation" to a single Light Zone.

[image]

T-Troll, 2025-03-06:
@amarildolsouza 
> Using manual lights setup to find correct IDs for this lights?

Yes.

@dc-IV-AW 

Some mech keyboards did not have a light under spacebar (it was checked carefully for m18r1, for example). Sad, but true.
Non-mech have.